sanni/cartreader #4: Satellaview Memory Pack Support

LuigiBlood, 2016-11-11:
We're quite in need for a better & cheaper dumper for Satellaview Memory Packs.

I've looked at the source code and it seems trivial to do, though I do not have the hardware to do all that but I wanted to talk to you about it.

I can provide where the Memory Pack is mapped on most cartridges that supports them, and also SRAM for BS-X carts.

sanni, 2016-11-12:
Skaman, a user from the Arduino and nesdev forums, has made a fork that specializes on the SNES.
It adds support for the following carts:
- Sufami Turbo Cart Support incl. SRAM  
- BS Satellaview Cart Support incl. SRAM & Memory Pack                                                          
- S-RTC Cart Support (Daikaijuu Monogatari 2) incl. SRAM & RTC                                    
- CX4 Cart Support (Mega Man/Rockman X2/X3)                                   
- Seta DSP (ST010/011/018) Cart Support incl. SRAM                                                              
- SPC7110 Cart Support (FEOEZ, MDH, SPL4) incl. SRAM & RTC

Afaik it has not been publicly released yet but he has asked for beta testers.

LuigiBlood, 2016-11-12:
Oh, I was originally planning to do it myself but if someone else did it I guess I may as well talk to skaman. Thanks for letting me know.

closed 2016-11-12 by LuigiBlood

borti4938, 2016-11-14:
I would like test it. How can I contact him? (I'm not registered in either mentioned forums and don't want to)

sanni, 2016-11-14:
You can also get in touch with him over the assemblergames forum, last reply in this thread: http://assemblergames.com/l/threads/snes-sfc-cart-dump-help-needed-cx4-chip.61547/#post-891377