sanni/cartreader #7: PCB does not convert logic level of SD card

nvsofts, 2017-03-13:
cartreader PCB uses "LC Studio" SD card module.
This module does not convert logic level of SD card.
This will be damaged or caused unexpected behavior to SD card.

I used this module to fix the problem.

microSD card module (Parts no: MCSD125A)
http://www.aitendo.com/product/10192 (Sorry, this page is Japanese)

The module uses SN74LV125A buffer IC to convert logic level.
However, the module does not contain 3.3V pin.
Therefore I applied small hack to the module (desolder AMS1117, short 5V-3.3V solder pad).

nvsofts, 2017-03-13:
similar item found in US Amazon
https://www.amazon.com/dp/B00SL0QWDU/

tamanegitaro, 2017-05-30:
nvsoft is correct.
Level shifter is not implemented on LC studio SD card module if the schematic in this thread is correct:
https://forum.arduino.cc/index.php?topic=369079.0
So you need to use level shifter for CS, SCK, MOSI, MISO pins or use microsd card adapter nvsoft provided.
I don't think you need to remove AMS1117 though because microsd card adapter also support 3.3V.
You don't need to remove AMS1117.

So best solution is to use microsd adapter and connect 5V to VCC on microsd adapter. So all CS, SCK, MOSI, MISO, and VCC will be 5V in 5V mode and be 3V in 3V mode.

sanni, 2017-07-10:
I added an option to install the suggested microSD module to the pcb. Since the cartreader shield already has an AMS1117 I desoldered it from the microSD module and shorted the 5V-3.3V solder pads on the AMS1117's footprint.    
    
But be aware that from my tests microSD cards are twice as slow as standard sized cards with the sdfat library I'm using for the cart reader. I don't know why though.

tamanegitaro, 2017-08-02:
Ahh my bad. Desoldering AMS1117 and connect 3.3V is best way to get around.

Supplying 3.3V to AMS1117 on micro SD card module is not enough because AMS1117 dropout voltage is 1.1V.
Which means voltage of sd card's signal line will become 2.2V and SD card require 2.7V-3.6V as minimum to maximum, respectively.
Another option will be to supply 5V (from slide switch) to AMS1117 on microsd module so I'm gong to try that.

closed 2018-04-21 by sanni