sanni/cartreader #14: Buy Pre-built Cart Reader?

iancourt, 2018-03-23:
Is there anyone out there offering for sale pre-built cartridge readers?

I don't particularly have the time, nor the SMT soldering equipment and skills, to build one myself, as much as I would like to?

bparker06, 2018-04-07:
Pretty sure these things would sell very well even if priced in the couple hundred range.

sanni, 2018-04-07:
There is no way I would be able to sell my Cart Reader commercially thanks to the need of FCC and CE certifications and many other regulations. All I can do is post my project here on github with the goal to form a community around it. 

The total parts cost for one Cart Reader are less than $50 and it takes maybe an hour to assemble. So my hope is that in the future able members of said community will contribute their soldering skills for free to this project and make the Cart Reader available for everyone at material cost. 

My suggestion is to either find a maker space in your area or visit the [topic on the Arduino forum](http://forum.arduino.cc/index.php?topic=158974.new#new) to see if someone can help you out.

Just be aware that anyone trying to sell you a Cart Reader for more than the material cost is doing so strictly against my will since what I don't want is people seeing this as an opportunity to make a quick buck by selling it for profit.

closed 2018-04-21 by sanni

bparker06, 2018-04-21:
FCC certifications are not required for sub-assemblies. If you just sell this as a shield, there is no issue.

And honestly, if you don't want people to sell this, then it probably shouldn't be free/open in the first place.

People are going to sell it anyway... would you rather them get all the money, or you?