sanni/cartreader #16: How to dump N64 Controller Paks / "Memory Cards"?

drtrigon, 2018-04-23:
I was looking into dumping my old N64 savegames from Controller Paks to file. So I had a look into https://github.com/sanni/cartreader/wiki/Reading-a-Controller-Pak but there is no info at all on **how to connect the N64 controller** - e.g. what is the "controller connector"? By browsing through the repo I could figure out that PH4 (Arduino Mega Pin 7) is the "Controller Data Pin" [1, 2]. And I can guess from the wiki that the controller needs 3.3V, but I cannot find from the scheme [3] where I am supposed to connect it and how. The best hint so far is this thread on "Reading a Controller Pak with your PC" [4].

[1] https://github.com/sanni/cartreader/blob/master/Cart_Reader/N64.ino#L346
[2] https://github.com/sanni/cartreader/tree/master/extras/controllertest
[3] https://raw.githubusercontent.com/sanni/cartreader/master/extras/pcb/cartreader_schematics.png
[4] http://s9.zetaboards.com/Nintendo_64_Forever/topic/7465098/1/

Finally I would like to use an Arduino Uno or may be an Arduino Yun (with SD card slot) to connect my N64 controller and dump my savegames from Controller Pak to file and store them on PC, respective use them in an emulator.

So my feature request is to update the wiki page with all info needed to properly connect the N64 controller to the cartreader. Optionally it would be great if you could also elaborate on how to connect and dump the controller pak using an Arduino Yun and on how to modify your source in order to do so.

sanni, 2018-04-23:
/**********************************************************************************
                  Nintendo 64 Controller Pak Reader for Arduino

   Author:           sanni
   Date:             2016-03-16
   Version:          V1

   SD  lib:          https://github.com/greiman/SdFat

   Thanks to:
   Andrew Brown/Peter Den Hartog - N64 send/get functions
   Shaun Taylor - address/data CRC functions

**********************************************************************************/

/******************************************
   Pinout
******************************************/
//N64 Controller 3.3V to Arduino 3.3V
//N64 Controller Data to Arduino Uno Pin 2
//N64 Controller GND to Arduino GND
//1K Resistor from Arduino Pin 2 to 3.3V

/******************************************
   Libraries
 *****************************************/
// SD Card
#include <SdFat.h>
#include <SdFatUtil.h>
#define chipSelectPin 10
SdFat sd;
SdFile myFile;

// AVR Eeprom
#include <EEPROM.h>
#include "EEPROMAnything.h"

/******************************************
  Defines
 *****************************************/
// Pin for controller data
#define N64_PIN 2

// these two macros set arduino pin 2 to input or output, which with an
// external 1K pull-up resistor to the 3.3V rail, is like pulling it high or
// low.  These operations translate to 1 op code, which takes 2 cycles
#define N64_HIGH DDRD &= ~0x04
#define N64_LOW DDRD |= 0x04
#define N64_QUERY (PIND & 0x04)

/******************************************
   Variables
 *****************************************/
// 256 bits of received Controller data
char N64_raw_dump[257];
// Array that holds one Controller Pak block of 32 bytes
byte myBlock[33];

//remember folder number to create a new folder for every save
int foldern;
char fileName[26];

// Variable to count errors
unsigned long writeErrors;

// For incoming serial data
int incomingByte;

// Array that holds the data
byte sdBuffer[32];
[... 460 lines cut ...]
  }

  // Read the incoming byte:
  incomingByte = Serial.read();

  // Execute user choice
  switch (incomingByte) {
    case 48:
      Serial.println(F("Reading Controller Pak"));
      readMPK();
      break;

    case 49:
      Serial.println(F("Writing Controller Pak"));
      getfilename();
      writeMPK();
      verifyMPK();
      break;
  }
  Serial.println("");
}

/******************************************
  End of File
******************************************/

sanni, 2018-04-23:
Updated the Wiki to make the operation more clearly. But keep in mind my Wiki assumes that you have build one of my Cart Readers. 

Additionally I have posted the above source code to dump Controller Paks with an Arduino Uno. Connect like so:     

- N64 Controller 3.3V to Arduino 3.3V
- N64 Controller Data to Arduino Uno Pin 2
- N64 Controller GND to Arduino GND
- 1K Resistor from Arduino Pin 2 to 3.3V

[image]

[image]

You also need to connect a standard Arduino SD card module and then just start the Serial Monitor.

[image]

drtrigon, 2018-04-25:
Wow! Thanks a lot for the wiki update! And the huge post providing all the code and docu I needed! Now I found the connector in the lower right corner of the schematic - and on the pcb also.

I took you code (thanks again for providing!) and adopted it for Arduino Yun using an USB stick (instead of SD card). You can find this code at https://github.com/drtrigon/sketchbook/blob/master/Yun_N64_Controller_Pak_Reader/Yun_N64_Controller_Pak_Reader.ino

The code compiles and runs fine but it does not work; it reads 0x00 all the time. The files contain zeros only. So ... what am I doing wrong? Is there a simple method to verify correct communication to the controller? (I noticed the read progresses even if not controller is connected at all) How can I be sure the connection is ok? Can you give me any hints?

drtrigon, 2018-04-25:
Ok, figured out how to check the communication to the controller - I added the following to the init() function:
```
  // Check/Test communication with N64 Controller
  Serial.print(F("Press any button on the N64 Controller to continue..."));
  while (button == "N/A") {
    delay(200);
    get_button();
  }
  Serial.println(button);
```
Now I have to solve the issue that I seem not to be able to make 3 simple wire connections, as I am not able to detect any key press...

sanni, 2018-04-25:
The ATmega32u4 chip used in the Arduino Yun probably behaves different compared to the ATmega328 in the Arduino Uno. You need to hook up a logic analyzer to make sure all the timings of the N64 controller protocol are still upheld and if not make the necessary changes to the code.

drtrigon, 2018-04-26:
I changed the code temporarily not to use SD card and test for button presses only. So I uploaded this code to an Arduino Uno - no success; no button press detected.
I opened the N64 Controller chasing and connected to the 3 pins (GND, VCC, DATA) directly - no success; no button press.
I connected a scope to figure out whether the Controller works ... but I could not work out how to test this - I do see some pulses but they come from the Arduino side only (go away as soon as I disconnect the Arduino).

sanni, 2018-04-26:
Not sure if it helps but:
- This is the project where the N64 controller protocol code initially came from: http://www.instructables.com/id/Use-an-Arduino-with-an-N64-controller/
- Make sure that your controller is an official Nintendo one and not a china clone.
- Here is an image of my build using an Arduino Uno, a standard sd shield, 1k resistor and a controller plug:
[image]

drtrigon, 2018-04-27:
... actually ... it helped indeed! Thanks a lot! Here my findings:

1. (finally) figured out how to connect the controller (using N64_Arduino)
2. figured out that **the code is incompatible with Yun/ATmega32u4** (using N64_Arduino)
3. thus code adopted from Yun to Uno: https://github.com/drtrigon/sketchbook/blob/master/Uno_N64_Controller_Pak_Reader/Uno_N64_Controller_Pak_Reader.ino
4. **add controller init procedure to ```setup()```** (missing in N64.ino ?): https://github.com/drtrigon/sketchbook/commit/ea4b8ec793f5b72a8efcbaf833f17171c018c5b7#diff-0ca416eab8bd379613acff558094346fR204
5. changed ```get_button()``` - do not check for eq 16 but neq 0
6. read output to serial console if ENABLE_SD not set
7. and add test option "2" in menu - use RESET to exit

So it's as you assumed, the code is not compatible with Yun/ATmega32u4 (2) and I think it has the bug of not initializing the controller in the beginning (4) - after I added this, it started working. Now the Read Menu option does read non-zero values which looks promising - now I have to connect my DIY SD Card reader and then inspect the data with https://rawgit.com/bryc/mempak/master/index.html - I'll report back...

drtrigon, 2018-04-27:
The picture your provided shows the SD Card chip select (CS) going to pin 4, but in the code we have ```#define chipSelectPin 10```. The other pins are not mentioned in the code at all... is this correct?

drtrigon, 2018-04-28:
I am fighting with the SD Card now ... ```sd.mkdir("MPK", true);``` does not create the folder and opening a file for writing afterwards fails. (it's a 16GB card with 2GB FAT16 partition)

sanni, 2018-04-28:
My bad, updated the picture to one that connects SS to pin 10. Fat32 should work fine. 
You might have to change the speed to half:
if (!sd.begin(chipSelectPin, SPI_FULL_SPEED)) {
to:
if (!sd.begin(chipSelectPin, SPI_HALF_SPEED)) {

drtrigon, 2018-04-28:
Thanks for the hint - tried SPI_HALF_SPEED now but gives the same result, cannot create the directory nor the EEPROM file. How did you partition and format the SD Card?

Meanwhile I used another way to get my data:
1. $ sudo miniterm.py /dev/ttyACM0 | tee n64-controllerpak-01.log
    * in the menu chose '0' to dump to console
2. open .log in text-editor (e.g. kate) and remove all lines except the ones containing the hex dump and store as it .hex - needs to be 1024 lines
3. $ xxd -r -p n64-controllerpak-01.hex n64-controllerpak-01.mpk
4. $ ls -la n64-controllerpak-01.mpk
    * check size needs to be 32768 and compare the .hex with .mpk using e.g. mc
    * finally inspect it on: https://rawgit.com/bryc/mempak/master/index.html

As this works for me now I was able to read an original Pak and the first bank of a Clone! Thanks a lot for you help and support! The code is in my repo as mentioned before.

However what i am not able to do is read other banks (than the 1st) on the Clone [1, 2]; after I pushed the button to switch it starts reading zeros only - but the Controller works (button presses are recognized). When I power-cycle the system, it reads the first bank again - no idea...

[1] https://i.ebayimg.com/images/g/8aUAAOSwPzhaPS-K/s-l300.jpg
[2] http://bordersdown.net/threads/95349-N64-Guillemot-Memory-64-DLX

drtrigon, 2018-04-28:
Ok, figured it out have to use https://github.com/greiman/SdFat/blob/master/examples/SdFormatter/SdFormatter.ino to format the SD Card, otherwise I did not work for me.

Only the question how to read additional memory banks from the Clone is open.