sanni/cartreader #19: Old PCB wiki

vogelfreiheit, 2018-05-30:
I have the earlier PCB (with spares I can sell, in black), but can't find any older wiki pages to build those anymore and I never saved it. Some steps definitely differ.

vogelfreiheit, 2018-05-30:
Also: 3d printed parts for the older SNES+N64 model missing :(

sanni, 2018-05-30:
Old version:
PCB and 3d printed stuff: https://github.com/sanni/cartreader/tree/09d122369fb5bbec1bffa3a6568dc0b8f87921db/pcb
Build guide: https://github.com/sanni/cartreader/wiki/PCB-Build-guide
Parts list: https://github.com/sanni/cartreader/wiki/Needed-Parts/26d82ed870367b93ec197ab153b63f9f27a8f25f
User Guide: https://github.com/sanni/cartreader/wiki/Setup-and-user-guide
Overview: https://github.com/sanni/cartreader/wiki/Cart-Reader-SNES-and-N64
Schematics: https://github.com/sanni/cartreader/blob/2c2eb0e8bd7ab05858fb0bab12f063c20793efb6/pcb/cartreader_schematics.png

closed 2018-06-06 by sanni