sanni/cartreader #20: pcb question

darleiv, 2018-06-01:
that blue board that solder to 29L the adapter maskrom you do not have gerber and esquematic? I'm trying to mount it but I did not understand the OE and A20 jumper - 
where are the jumper oe and a21 connected? Would you have the schematic or gerbber brd?

darleiv, 2018-06-01:
https://www.sendspace.com/file/82bln6

sanni, 2018-06-01:
New design to flash 29L3211 looks like this: https://github.com/sanni/cartreader/wiki/Programming-a-29LV160-flashrom

It consists out of the flash_adapter pcb found [here](https://github.com/sanni/cartreader/tree/master/extras/pcb) and [this adapter](https://www.aliexpress.com/item/SOP44-to-DIP44-PSOP44-DIP44-SOP44-SOIC44-SA638-B006-IC-IC-socket-Programmer-adapter-Socket-High/691017928.html) from aliexpress.

The old design in your picture is not available anymore because it did not work that well.

closed 2018-06-06 by sanni