sanni/cartreader #26: Status of firmware compatibility with SNES and N64 old version

vogelfreiheit, 2018-10-09:
Hi,

I have built an older version of the hardware, with N64 and SNES slots. Does the current firmware support that version? Or I need to modify any pin settings or such?

sanni, 2018-10-09:
It's 100% compatible with the newest firmware. I'm still using one of the earlier Cart Reader PCBs myself.

vogelfreiheit, 2018-10-09:
Thanks man! I might consider selling some kits or bare PCBs if you are OK with that. I just made too many. I can also send you some. Got photos of the silkscreening and all. I will flash the  board now and check back.

BTW I populated the capacitor at the USB input. I never saw it mentioned anywhere in the guides but it shouldn't hurt (the electrolytic).

vogelfreiheit, 2018-10-09:
One issue I encountered with some N64 carts: flashram unknown is reported. I have the external clock ref enabled and plugged in through a pin header. What could possibly cause this?

vogelfreiheit, 2018-10-09:
Re-tested with both Arduino-sourced power and using the port on the shield. I'm using 3.3V, no EEP and EEP and the clock settings turned off. No improvement there.

sanni, 2018-10-10:
Have you tested dumping the rom of the cartridge to make sure everything works with the N64 cart slot? Also try to write a save game to a cartridge with the sram save type to verify that writing to a N64 cartridge in general works so you can rule out a cold solder joint.

vogelfreiheit, 2018-10-12:
I will upload a patch later, I had to add support for two cartridges that use one of the 'alternative' configurations (they can either have a MN63F81MPN or 29L1100KC) that identify as different ICs, but are still supported by the present code for 29L1100KC-15B0.

The codes were 0x84 and 0x8e if I recall correctly. The hardware itself presents no issues. I get more problems out of the GBA/GBC slot adapter+cartridge setup than anything else. N64 carts and anything with direct slot plugging works fine. 

It might be worth considering adding a PCB that takes a salvaged GBC cart slot, which are arguably the best for that system (no pin wear and no issues with the cart moving slightly and causing contact problems).

sanni, 2018-10-13:
Yes that's the reason why the current PCB design has both the MD and GB slot directly on the main PCB without any adapters.

[image]

[image]

closed 2018-11-20 by sanni