sanni/cartreader #27: Read Megadrive 32mbits

seven-sam, 2018-10-13:
Hello,

I tried to dump Super Street Fighter II, but under an emulator it is not readable, I try other games like sonic, aladin and those works well ...

an idea ?

Do you plan to write megadrive cartridges?

Thx

Sam

sanni, 2018-10-14:
Super Street Fighter II is one of the few Mega Drive games that have a mapper chip on the PCB. Currently I have not written the support for any mapper chips or any other special Mega Drive cartridges.

I have some code for writing Mega Drive repros with MX29F1610 flash chip but this is not added yet to the main branch. 

What currently is available is the code to write 27C322 eproms which are often used for Mega Drive reproductions, however I'm still waiting on the 27C322 flash adapter PCB.

As for re-writing Chinese Mega Drive repros from ebay or Aliexpress I have not looked into that so far and since the Chinese seem to use a different flash chip every few weeksi for their repros I don't think it's worth the time especially considering an Everdrive MD clone can be bought for less than $25.

seven-sam, 2018-10-15:
thank you for your reply ;)
Ok I understand, it's true I also have an everdrive, but it's to do it to be able to do it: p

closed 2019-12-27 by sanni