sanni/cartreader #57: Possibly update BOM for OLED

birdybro, 2020-11-19:
The 0.96" 4 Pin IIC OLED your BOM suggests is now on 2 and a half months being shipped to me. I just said screw it and have ordered one from Amazon for only 7 bucks from HiLetGo, so just a heads up as other people may run into issues from "Win Win" supply chain and shipping. Ordered Sep 3rd and it hasn't even reached US Customs yet for some reason.

https://www.amazon.com/gp/product/B06XRBTBTB/

Here's the alternative you could mention. Only increases price by a few bucks.

sanni, 2020-11-19:
Thanks for your input. I have linked to this issue from the parts list so that everyone can share their favorite sellers here.

sanni, 2020-12-08:
Moved to Discussions

closed 2020-12-08 by sanni