sanni/cartreader #68: Checksum fail trying to write .sav file

rafaliyo, 2021-01-08:
Some users and I had problems trying to write a .sav file into a flashcart. You can play and save but once you try to read/write the file you cant

The flashcart has a FRAM chip to replace the original SRAM

We have tested with other reader and it works good. Maybe its because the sanni tries to write in save in a SRAM chip when its a FRAM? Could it be a firmware issue?

[image]

Thanks in advance

darleiv, 2021-01-09:
works normally, if you want to check sun ok do a fram dump and record. you used sram save

sanni, 2021-01-09:
You probably have to add more delays to the current code since the FRAM chip most likely is slower than the original SRAM.    
Functions related to SRAM reading/writing in gb.ino:    
readByteSRAM_GB
writeByte_GB
writeSRAM_GB

rafaliyo, 2021-01-09:
Talking from the unknowledge, it is difficult to do it? I dont know anything about coding :( . I got the sanni working because the guide is very clear even for me

AndehX, 2021-01-10:
Same issue here, although I have no coding experience so I'm unable to fix it.  Do you think you could quickly adjust it sanni?

darleiv, 2021-01-10:
there is no need to add delays. you don't need to change the code. just save and even with the error it works 100% save. and if you dump after resave it will not have a checksun error. and the type of file that is in trouble - no use for delays. the error is in the proper save not compatible in size with the fram. understood?

darleiv, 2021-01-10:
I added mips and delays and it didn't solve anything. the problem is in the save file with the wrong size, after correcting this checksun 100%

darleiv, 2021-01-10:
> You probably have to add more delays to the current code since the FRAM chip most likely is slower than the original SRAM.
> Functions related to SRAM reading/writing in gb.ino:
> readByteSRAM_GB
> writeByte_GB
> writeSRAM_GB

mirror save to the exact sram size which is 256k and resolved 8k+8k+8k+8k - 32k = => FRAM 1808

AndehX, 2021-01-10:
I don't understand what you mean.  Whenever I write a save, it doesn't work.  I write saves for Pokemon, and when I load game, all I get is "New Game"

darleiv, 2021-01-10:
got it. see if the sanni is correct at 3.3v or 5v depending on the frame it changes. I was talking about super nintendo with fram. , if you use 3.3 volts in a 5v fram not saved

darleiv, 2021-01-10:
> I don't understand what you mean. Whenever I write a save, it doesn't work. I write saves for Pokemon, and when I load game, all I get is "New Game"

got it. see if the sanni is correct at 3.3v or 5v depending on the frame it changes. I was talking about super nintendo with fram. , if you use 3.3 volts in a 5v fram not saved

AndehX, 2021-01-10:
I'm using the correct voltage, lol.

sanni, 2021-01-11:
Here is what I would do:      
1) Load the ROM in an emulator, start a new game, save as soon as possible and open the emulator save file with an hex editor like Nexus HxD     
2) Play the cartridge in a GameBoy, start a new game, do exactly the same steps as in the emulator and save as soon as possible, dump the save and open the dumped save file with an hex editor like Nexus HxD      
3) Compare both saves to see if they are the same or completely different, if they are mostly the same then reading the FRAM works if not add a second line of  "__asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");" everywhere in byte readByteSRAM_GB(word myAddress)       
4) Once you can dump the save correctly proceed with modifiying the writeByte_GB the same way until it starts working.      
5) If it still doesn't work I'm afraid you will have to read the datasheet of your FRAM chip to get the exact timings needed.

rafaliyo, 2021-01-11:
As I said before (I dont know about coding) I have what you say in step 1-2 sanni. 

I have saved in my original cartridge of pokemon blue and I have open the game in visual boy advance and save it in the same point (once I sart the game)

I have put both files in HXD programm (I dont know if I am doing it good) and they looks similiar until offset 00000490 where is file starts to change. Is it possible to upload the files here?

darleiv, 2021-01-11:
> As I said before (I dont know about coding) I have what you say in step 1-2 sanni.
>
> I have saved in my original cartridge of pokemon blue and I have open the game in visual boy advance and save it [...]

only one doubts the checksun error. but the save does not work the same even with this error ??? as I reported I had a snes checksun error and it was because the fram has 256k not 64k there and only dumpar and rewrite the check sun ok

rafaliyo, 2021-01-12:
> only one doubts the checksun error. but the save does not work the same even with this error ??? as I reported I had a snes checksun error and it was because the fram has 256k not 64k there and only [...]

No, the dump does not work.

sanni, 2021-01-17:
> . Is it possible to upload the files here?

You can just drag and drop the files into the message box here to upload them, just have to zip them beforehand.

rafaliyo, 2021-01-19:
Here are the files:

[saves.zip](https://github.com/sanni/cartreader/files/5835395/saves.zip)

sanni, 2021-01-19:
That actually looks quite good since the difference is only in the area where the emulator has 0xFF meaning that the game probably doesn't clear the whole SRAM with 0xFF before saving.

Try writing this random save file, extract first ofc, and then read the save again, without playing the game. This will show me what goes wrong when writing.

[random.zip](https://github.com/sanni/cartreader/files/5836513/random.zip)

Also, what game is it? And what is written on the new FRAM chip?

rafaliyo, 2021-01-20:
> 
> 
> That actually looks quite good since the difference is only in the area where the emulator has 0xFF meaning that the game probably doesn't clear the whole SRAM with 0xFF before saving. So I would assume that reading works.
> 
> Try writing this save file filled with random characters, extract first ofc, and then read the save again, without playing the game. Then upload it here again. This will show me what goes wrong when writing.
> 
> [random.zip](https://github.com/sanni/cartreader/files/5836513/random.zip)
> 
> Also, what game is it? And what is the parts number of the new FRAM chip, so I can check the datasheet?

Its pokemon blue, and the FRAM chip is FM18W08-SG

About the save you have put, you mean write it in the original cartridge with your cartreader?

I would have to write it, load, save the game and read it again to see what change?

sanni, 2021-01-20:
Write the save to the game cartridge with the Cart Reader and then immediately read it back again with the Cart Reader. Don't put the game in a GameBoy.

rafaliyo, 2021-01-20:
If I put your random save into the SD and try to flash it with the cart reader I get the same error but it generate a save file, this one:
[POKEMONB.zip](https://github.com/sanni/cartreader/files/5842050/POKEMONB.zip)

sanni, 2021-01-20:
Ok the random 8KB file got written perfectly. Next try this 32KB random save. 

[rand32KB.zip](https://github.com/sanni/cartreader/files/5842094/rand32KB.zip)

rafaliyo, 2021-01-20:
with that one I get no error. This is the save file I read after I write you sent
[POKEMONB32.zip](https://github.com/sanni/cartreader/files/5842243/POKEMONB32.zip)

sanni, 2021-01-20:
Ok, one more test. This is a save file I got from the internet. Flash it and then play the game in the GameBoy. The player should be named Ash and he should have a Pokemon called Mew.

[POKEMEW.zip](https://github.com/sanni/cartreader/files/5842254/POKEMEW.zip)

rafaliyo, 2021-01-20:
Yes, it has completed the pokedex and it has only mew in the team

sanni, 2021-01-26:
Ok then let's start fresh before it gets too confusing.

Start a new Pokemon Blue game on the flashcart with the FRAM chip and save and then dump the save with the Cart Reader. Do you get a 32KB file that has the same content as if you had done the same on the original Pokemon Blue cartridge?

If yes, write the rand32KB save to the flashcart with the FRAM chip and immediately dump it again. Compare the dump with the rand32KB save. How does it differ?

rafaliyo, 2021-01-28:
Hi

If I start a game in the flashcart, save it and compare it with a save from the original cartridge in the same point and I put both files in HXD programm, (I dont know if I am doing it good) they looks similiar until offset 00000490 where is file starts to change

If I write the rand32KB save to the flashcart (getting the error yet) and I back up it immediately and I compare them after, they looks different. Here are the files:

FCS= save from flashcart as soon as I can save in the game
OS= save from original cartridge as soon as I can save in the game
R32= save from flashcart once I have written rand32KB save and dump it again

[test saves.zip](https://github.com/sanni/cartreader/files/5886694/test.saves.zip)

sanni, 2021-01-31:
Ok,
so the Cart Reader fails to write the first 64bytes, then skips one byte and writes the rest almost perfectly with the occasional write error, although everything is shifted to the left by that one byte it skipped.

So I would try adding delays and test writing the rand32 save again to see if anything has changed.

In GB.ino line 338 I would just add a few more nops
`
void writeByte_GB(int myAddress, uint8_t myData) {
  PORTF = myAddress & 0xFF;
  PORTK = (myAddress >> 8) & 0xFF;
  PORTC = myData;

  // Arduino running at 16Mhz -> one nop = 62.5ns
  // Wait till output is stable
  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t");

  // Pull WR(PH5) low
  PORTH &= ~(1 << 5);

  // Leave WE low for at least 60ns
  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t");

  // Pull WR(PH5) HIGH
  PORTH |= (1 << 5);

  // Leave WE high for at least 50ns
  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t");
}
`

And then in line 655 I would also add nops like so:
`
// Write RAM
        for (uint16_t sramAddress = 0xA000; sramAddress <= sramEndAddress; sramAddress++) {
          // Pull CS(PH3) LOW
          PORTH &= ~(1 << 3);
         __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t");
          // Write to RAM
          writeByte_GB(sramAddress, myFile.read());
          __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t");
          // Pull CS(PH3) HIGH
          PORTH |= (1 << 3) ;
          __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t""nop\n\t");
        }
`

This is ofc just trial and error. I tested it with an original SRAM and the code still works, question is are those delays enough to make the FRAM chip work.

rafaliyo, 2021-02-02:
Ok then I just would have to open that file in Arduino and upload it right?

sanni, 2021-02-02:
You need to download and extract V5.5_Portable.zip from here https://github.com/sanni/cartreader/releases/download/V5.5/V5.5_Portable.zip
And then copy the GB.ino extracted from GB.zip in above's post to \V5.5_Portable\Arduino IDE Portable\portable\sketchbook\Cart_Reader\ and overwrite the existing file. Then launch \V5.5_Portable\Arduino IDE Portable\arduino.exe and flash the sketch like explained here https://github.com/sanni/cartreader/wiki/How-to-flash-the-Arduino

rafaliyo, 2021-02-03:
It stills get the same error (trying to write rand32 save):
SRAM writing finished
Error: 362640 bytes
did not verify

closed 2021-04-06 by sanni