sanni/cartreader #111: Parts in the excel for LCSC should be updated

birdybro, 2021-03-30:
[image]

Maybe suitable replacements (could be wrong):
https://lcsc.com/product-detail/MICROCHIP_Microchip-Tech-PIC12F675-I-SN_C9730.html - Only difference I see is the 675 supports analog to digital conversion but is backwards compatible with the 629. Is that correct? It's also slightly more expensive but not by much.

[image]

Suitable replacements for similar or lower prices:
https://lcsc.com/products/High-Precision-Low-TCR-SMD-Resistors_911.html?attribute=%7B%22Resistance%20(Ohms)%22%3A%5B%221K%22%5D%2C%22Tolerance%22%3A%5B%22%C2%B10.1%25%22%5D%2C%22Power%20(Watts)%22%3A%5B%221%2F8W%22%5D%7D&package=0805

[image]

Suitable replacements for similar or lower prices:  
https://lcsc.com/products/Chip-Resistor-Surface-Mount_439.html?attribute=%7B%22Resistance%22%3A%5B%2210k%CE%A9%22%5D%2C%22Tolerance%22%3A%5B%22%C2%B15%25%22%5D%2C%22Power(Watts)%22%3A%5B%220.125W%22%5D%2C%22Temperature%20Coefficient%22%3A%5B%22%C2%B1100ppm%2F%E2%84%83%22%5D%7D&package=0805

Thank you.

HighMans, 2021-03-30:
Do the 1k resistors need to be +/- .1% tol? The ebay listing for bulk order shows that they're +/- 5% tol.

sanni, 2021-03-30:
Thanks for the resistor links. I have updated the file. The PIC probably is only out of stock temporarily since it still says In Production on microchip.com

closed 2021-04-01 by birdybro