sanni/cartreader #117: Fram Mod Not Recovering Save

Coaleb2, 2021-04-06:
I did the Gameboy Immortalizer save mod. Where you swap out the sram for fram. This is the link to the kit https://www.tindie.com/products/obskyr/save-immortalizer-kit-for-game-boy/. I got the original save dumped but when I recover it it gives checksum error. It won't even rewrite a new save over a cleared save. What could I be doing wrong or is this cart dumper incompatiable with this mod. The fram chip is Ramtron FM18W08-S0

sanni, 2021-04-06:
You can try replacing the [GB.ino file with the one attached here](https://github.com/sanni/cartreader/files/6263129/GB.zip) and reflash the Arduino. I have not tested it since don't have any Ramtron chips though. So it might still not work.

More info for the tech-savvy:
- [Ramtron datasheet](https://github.com/sanni/cartreader/files/6263159/001-86207_FM18W08_256-Kbit_32_K_8_Wide_Voltage_Byt-476756.zip)
- [Write/Read capture](https://github.com/sanni/cartreader/files/6263148/write_read_gameboy_sram.zip) of the changed GB.ino from [Logic Analyzer](https://downloads.saleae.com/logic2/Logic+Setup+2.3.24-master.exe).

Coaleb2, 2021-04-06:
That did it. Thank you so much. I can now continue my LOZ Oracle of ages save.

sanni, 2021-04-06:
Thanks for testing.

closed 2021-04-06 by sanni