sanni/cartreader #121: Problem when reading save from Nintendo 64 EEPROM cartidges

parski, 2021-04-12:
I've been dumping Nintendo 64 SRAM games without issue but now attempting to dump EEPROM save files it appears to crash. These are my steps:

DIPS:

```
+-------+-------+
| 3V/5V | 3V  < |
+-------+-------+
| EEP   | ON  > |
+-------+-------+
| CLK0  | OFF < |
+-------+-------+
| CLK1  | OFF < |
+-------+-------+
```

1. Turn the unit ON.
2. Select **Nintendo 64**
3. Select **Game Cartridge**

```
N64 Cartridge Info

Name: MARIOKART64
ID: NKTJ Size: 12MB
Save: 4K Eeprom
Version: 1.0

Press Button...
```

4. Press button
5. Select **Read Save**

LED flashes blue maybe five times in rapid succession, then LED is persistently off. Stuck on the screen is the following:

```
Reading Eep...
```

I've tried leaving the unit on for a good 20 minutes and I'm assuming something is wrong and that this is not an overnight operation. How can I troubleshoot this? As previously mentioned, dumping SRAM is not a problem.

parski, 2021-04-12:
When checking the SD card after an attemt it has created a directory structure like so:
```
/N64/SAVE/{game}/{integer}/
```
With one `{integer}` directory per attempt. These directories are empty.

sanni, 2021-04-13:
Make sure its an original cartridge and not a Chinese reproduction.
Reading the eeprom needs two connections, data and clock. Try resoldering all the points on the PCB involved and make sure the 1K resistor in the data line is connected properly too.

You can also use a multimeter to check all the traces.

This is the clock line on the PCB, it comes from the Arduino pin D16 at the top, goes through the CLK1 switch(switched to off) and then finally into the CLK pin on the N64 cart slot:
[image]

This is the data line, it goes from Arduino pin D7 straight into the DAT pin on the N64 cart slot, but is also connected to the EEP switch and through the EEP switch(switched to on) it goes to the 1K resistor to VCC:
[image]

If you got the N64 controller plug installed you can also test the DAT pin with a N64 controller, if the controller test works the DAT pin is ok. However the CLK pin is not used with the controller.

If all the physical connections are ok you can clean the cart edge connector of the game with a q-tip and some rubbing alcohol.
Lastly you can try another eeprom game to see if the issue maybe is specific to the particular eeprom chip in Mario Kart. I always wanted to rewrite the current eeprom code to utilise the clock generator to be more precise, but so far all the carts I tested worked with the current somewhat crude implementation that doesn't need a clock gen.

parski, 2021-04-14:
Thank you for your exceptional response.

I get continuity between D16 on the Arduino and the CLK pin in the Nintendo 64 cartridge slot with the CLK1 switch set to the OFF position, as you described.

I also get continuity between D7 on the Arduino and the DAT pin in the Nintendo 64 cartridge slot regardless if the EEP switch is set to ON or OFF.

I've tried several cartridges including a launch copy of Super Mario 64 so I'm certain they are not counterfit. 

I'll see if I can find an expansion pack and try to dumb saves via a controller since I'm less sure about the DAT pin as it involves a resistor and VCC.

I'll also grab a bottle of isopropyl alhohol and a few other EEPROM based games and report back.

sanni, 2021-04-14:
The EEP switch enables the 1K resistor on the DAT line. To test that, switch the Cart Readers On/OFF switch to OFF then use the multi-meter in Ohm mode and measure the resistance between the DAT pin and the VCC pin on the N64 cart slot. It should be around 1 kOhm when the EEP switch is set to ON.

[image]

parski, 2021-04-14:
I'm getting within reasonable tolerances the same 1kΩ I expect between DAT and VCC when the main power switch is set to OFF and EEP switch is set to on.
[image]

sanni, 2021-04-15:
I just added an alternative way to read/write the Eeprom if you have the Adafruit Clockgen module installed with V5.7:

In options.h change:
```
// Read N64 Eeprom with Adadruit clockgen, CLK1 switch needs to be switch to ON
//#define clockgen_installed
```

to: 
```
// Read N64 Eeprom with Adadruit clockgen, CLK1 switch needs to be switch to ON
#define clockgen_installed
```

Then setup the switches like this:    

[image]

parski, 2021-04-15:
I'm happy to report that your clockgen fix **solved the issue**. I've dumped the EEPROM saves of a PAL version of Super Mario 64 as well as an NTSC-J version of Mario Kart 64 successfully using what's currently the latest `master`.

Thank you for your kind help. This issue can be closed.

closed 2021-04-15 by sanni

sanni, 2021-04-15:
Glad I could help. 👍