sanni/cartreader #133: Dumping N64

Coaleb2, 2021-04-27:
Dumping Pokemon Stadium 2 and Conkers Bad Fur Day Will give a crc error. Cleaning the port and cleaning the pins did not fix the issue. I tried multiple dumps to no avail

sanni, 2021-04-27:
- If those games are reproductions then the CRC error is to be expected.

- If they are original then please check the voltage between those two points, it needs to be 3.3V when the switch is in 3V mode:

[image]

Other than that I can only think of the cart being seated not perfectly centered into the cart slot.

Coaleb2, 2021-04-27:
Okay mine is staying at 5v so that could be my problem. The rest of my games worked fine which is weird

sanni, 2021-04-27:
My Pokemon Stadium does the same. It needs 3.3V to dump. Older/smaller carts also work with 5V.

Make sure you followed the [build guide](https://github.com/sanni/cartreader/wiki/How-to-build-the-PCB) and did not forget to desolder the 5V regulator and the transistor from the Arduino board:
[image]

Coaleb2, 2021-04-27:
Yeah thats where I am confused. I did de-solder that on mine. I re-soldered the points with a 470uf capacitor added in

Coaleb2, 2021-04-27:
I see what I did wrong. I missed removing the transistor. That explains a lot. I must have added it back in when I was flashing the Arduino.

Coaleb2, 2021-04-27:
Once I removed it it instantly worked dumping one of the games I could not before. Thank you for the help. I would not have noticed that transistor being there

closed 2021-04-28 by sanni