sanni/cartreader #296: Gameboy dump issue

drpaneas, 2021-12-02:
I've just got mine (pre-assembled) but any Gameboy cartridge I try to dump is not recognized. No name, no rom, just some broken files. How can I troubleshoot what's wrong there?

sanni, 2021-12-02:
The most efficient way would be to contact your seller and ask for a replacement but if that is not an option or if you like the challenge of fixing it yourself then I found that the fastest way to troubleshoot the hardware is to compare a broken dump to a known good one of the same game in an hex editor and then try to see a pattern which points to a specific address or data line being the culprit. 

It's important to know that with the hardware V3 all slots are connected in parallel so the hardware defect could be on any of the slots and still have an effect on the Gameboy slot. This is something I changed with hardware V4.

To find the issue you should use a standard Game Boy (Color) game, not Game Boy Advance since GB(C) has an easier protocol where address lines and data lines are split.

Here is an example of this way to troubleshoot from the perspective of the N64 slot: https://forum.arduino.cc/t/rom-reader-for-super-nintendo-super-famicom-game-cartridges/155260/1270?u=sanni

And here the same issue but seen from the GB slot: https://forum.arduino.cc/t/rom-reader-for-super-nintendo-super-famicom-game-cartridges/155260/1272?u=sanni

So using different slot you can also crosscheck the issue. In this case the user had a bad solder connection on Arduino Pin A8.

Dakkaron, 2021-12-03:
Looks like the kickstarter mess is rolling in... If you can't get the device fixed by the seller, try to resolder every single solder connection on the board.

Androxilogin, 2021-12-03:
> Looks like the kickstarter mess is rolling in... If you can't get the device fixed by the seller, try to resolder every single solder connection on the board.

Unfortunately this is my problem. I used Kester solder rosin core. The flux apparently backfed into the switches and hardened even after vigorously cleaning the board and components afterwards. So it was working fine while in my hands then presented itself later on. We've been in contact about this issue and walking through it.