sanni/cartreader #313: Fram Mod gbc saves not dumping

Coaleb2, 2021-12-24:
I have the fram mod installed on my pokemon crystal. I can no longer dump the save from it. Is there a way to dump the save for v3 cart dumper? The Fram model number is Ramtron fm1808.

sanni, 2021-12-26:
The only thing I can think of is reducing the delays in the [readByteSRAM_GB function](https://github.com/sanni/cartreader/blob/master/Cart_Reader/GB.ino#L476)

````
byte readByteSRAM_GB(word myAddress) {
  PORTF = myAddress & 0xFF;
  PORTK = (myAddress >> 8) & 0xFF;

  // Pull CS(PH3) LOW
  PORTH &= ~(1 << 3);
  // Pull RD(PH6) LOW
  PORTH &= ~(1 << 6);

  __asm__("nop\n\t""nop\n\t");

  // Read
  byte tempByte = PINC;

  // Pull CS(PH3) HIGH
  PORTH |= (1 << 3);
  // Pull RD(PH6) HIGH
  PORTH |=  (1 << 6);

  return tempByte;
}

````

Did you just replace the SRAM chip with a FRAM chip or did you also add the extra parts needed when FRAM modding a MBC3 cartridge?

sanni, 2021-12-26:
Without knowing exactly what you modded I can only take guesses.
So let's assume you are using this mod.

[image]

It seems it use the CLK pin as CE for the FRAM. So you need to change some functions in GB.ino:

https://github.com/sanni/cartreader/blob/master/Cart_Reader/GB.ino#L319
````
  // Set Control Pins to Output RST(PH0) CLK(PH1) CS(PH3) WR(PH5) RD(PH6)
  DDRH |= (1 << 0) | (1 << 1) | (1 << 3) | (1 << 5) | (1 << 6);
  // Output a high signal on all pins, pins are active low therefore everything is disabled now
  PORTH |= (1 << 0) | (1 << 1) | (1 << 3) | (1 << 5) | (1 << 6);
````

https://github.com/sanni/cartreader/blob/master/Cart_Reader/GB.ino#L476
````
byte readByteSRAM_GB(word myAddress) {
  PORTF = myAddress & 0xFF;
  PORTK = (myAddress >> 8) & 0xFF;

  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");

  // Pull CS(PH3) CLK(PH1) LOW
  PORTH &= ~((1 << 3) | (1 << 1));
  // Pull RD(PH6) LOW
  PORTH &= ~(1 << 6);

  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");

  // Read
  byte tempByte = PINC;

  // Pull CS(PH3) CLK(PH1) HIGH
  PORTH |= (1 << 3) | (1 << 1);
  // Pull RD(PH6) HIGH
  PORTH |=  (1 << 6);

  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");

  return tempByte;
}
````

https://github.com/sanni/cartreader/blob/master/Cart_Reader/GB.ino#L502
````
void writeByteSRAM_GB(int myAddress, uint8_t myData) {
  PORTF = myAddress & 0xFF;
  PORTK = (myAddress >> 8) & 0xFF;
  PORTC = myData;

  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");

  // Pull CS(PH3) CLK(PH1) LOW
 PORTH &= ~((1 << 3) | (1 << 1));
  // Pull WR(PH5) low
  PORTH &= ~(1 << 5);

  // Leave WE low for at least 60ns
  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");

  // Pull WR(PH5) HIGH
  PORTH |= (1 << 5);
  // Pull CS(PH3) CLK(PH1) HIGH
  PORTH |= (1 << 3) | (1 << 1);

  // Leave WE high for at least 50ns
  __asm__("nop\n\t""nop\n\t""nop\n\t""nop\n\t");
}
````

Another idea would be removing some delays:

https://github.com/sanni/cartreader/blob/master/Cart_Reader/GB.ino#L476
````
byte readByteSRAM_GB(word myAddress) {
  PORTF = myAddress & 0xFF;
  PORTK = (myAddress >> 8) & 0xFF;

  // Pull CS(PH3) CLK(PH1) LOW
  PORTH &= ~((1 << 3) | (1 << 1));
  // Pull RD(PH6) LOW
  PORTH &= ~(1 << 6);

  __asm__("nop\n\t");

  // Read
  byte tempByte = PINC;

  // Pull CS(PH3) CLK(PH1) HIGH
  PORTH |= (1 << 3) | (1 << 1);
  // Pull RD(PH6) HIGH
  PORTH |=  (1 << 6);

  return tempByte;
}
````

Coaleb2, 2021-12-27:
Okay I am using this mod https://www.tindie.com/products/obskyr/save-immortalizer-kit-for-game-boy/ . I tested out what you told me to change. It did not allow my Pokemon Crystal to dump unfortunately

Coaleb2, 2021-12-27:
Okay that worked. Thank you again for your amazing device

closed 2021-12-29 by sanni