sanni/cartreader #321: SD Error after update to V7.3

muellerm1, 2022-01-23:
I have a HW3 cartreader, and I recently saw that its software is out of date, so I updated it to 7.3, and now it just says SD Error on startup. I have not been able to identify the cause, and I don't see the older versions on the Releases page anymore, so I'm stuck.

HW3 is defined in options.h

sanni, 2022-01-23:
You can find older releases archived here: https://github.com/sanni/cartreader/wiki/Cart-Reader-V3
They were removed from the releases tab because they don't work with HW4.
With the 6.0 release the SD library was updated to SdFat Version 2.
You can try reformating the card to exFAT which is now supported thanks to SdFat V2.

muellerm1, 2022-01-23:
Update:
Thanks for those links to the old releases. I've verified that V5.5 still works if I upload it, so at the very least, there's nothing to lose anymore. Are the changelogs visible anywhere?

Unfortunately, reformatting the SD card did not resolve the problem.

I've tested formatting the SD card as FAT32 (which is what it was originally) and exFAT, for each V5.5, V6.4, V6.6, and V7.3.
V5.5 works with FAT32, but gives an SD Error on exFAT.
V6.4 and V6.6 give an SD Error on both FAT32 and exFAT, but they display the mushroom logo on startup first.
V7.3 gives an SD Error on both FAT32 and exFAT, but the mushroom logo doesn't display. Sometimes garbage displays in its place, like static. But only sometimes.

Formatting the SD card as exFAT has not made a visible difference, except that it stops working on V5.5.

sanni, 2022-01-23:
Can you try a different SD card? Preferably 32GB or smaller and FAT32 formatted.
It could be that the new SdFat V2 library is not compatible with your SD card.

muellerm1, 2022-01-24:
Yep, that was it. The old card was only a few years old, but I guess that's all it takes. Thanks!

closed 2022-01-24 by muellerm1