sanni/cartreader #360: Xplorer64 not read.

Nes-player4Life, 2022-04-28:
Cheat devices, Xplorer64 (Xploder64) not read.

sanni, 2022-04-29:
What happens if you try to dump it?

Nes-player4Life, 2022-04-29:
Dumps an empty file. GPERROR. Thousands of lines of 0

sanni, 2022-04-29:
Can you make a picture of the Xplorer64 PCB (front and back)?

Nes-player4Life, 2022-05-01:
[image]
[image]
[image]
[image]
[image]
[image]

As you see some of the chips look like the gameshark chips.

sanni, 2022-05-04:
Maybe do a complete 64MB dump and then search for Xplorer64 in the file using a hex editor to locate where the bios starts. For example with the Gameshark the bios gets read starting from offset 0xC00000. It could be similar with the Xplorer64.

majorpbx, 2022-05-13:
I just got one of these carts and it did not dump as well. I haven't had time to troubleshoot but will see if I can get it to work.

majorpbx, 2022-05-13:
As as side note, I have successfully dumped GameShark v1.05-3.3, Pro & Action Replay versions, Passport & Plus verisons, and Equalizer. These all have the same if not similar ROM chips.

Nes-player4Life, 2022-05-14:
You dumped the passport with this device? Very cool. And the 1.0.5 gs that is a very early version sweet!

majorpbx, 2022-05-14:
I did. I cleaned Xplorer64 and it did not dump ☹️ and tried Super Mario 64 (USA) cart and got a blank dump with random data in it.

I also have a Game Killer to dump but this uses a completely new single chip. Uses a PIC16C558-04/S0 chip and looks to be much simpler logic board. I'm guessing for this Game Killer it might need a new method for dumping.

Nes-player4Life, 2022-05-15:
What version of xplorer64 do you have? I have the orange .0.6. E  i think it is.

majorpbx, 2022-05-16:
My version info is below. Press and hold Z and you should get something maybe?

Xplorer N64 Version
(C) 1997/9 FCD Release 1.000E
England

Build 1834
Compiled:
Mon Aug 15 12:10:59 BST 1999

closed 2023-08-18 by sanni