sanni/cartreader #378: Pocket SNES Cart RW - Clock generator not working!

Hanzel1985, 2022-06-15:
for some reason, with the new update, i can enable the adafruit clock generator, even if i configure in the main option part in the arduino programmer, can someone help¡

sanni, 2022-06-18:
Check that the SDA and SCL pins of the Clock Generator are connected to the Arduino.
The newer Cart Reader versions check if the Clock Gen is actually working and print and error if it doesn't.

Hanzel1985, 2022-06-20:
thanks for the reply sanni!, i see, but where i need to connect those pins??, im using the snes pocket cart rw that uses the arduino mega 2560 pro,

sanni, 2022-06-20:
I don't know how the Pocket SNES Cart RW is wired as it is not my design. Can you make a picture how the Clock Generator is connected?

closed 2022-06-24 by sanni