sanni/cartreader #379: 5V works, 3.3V does not

sanni, 2022-06-18:
### Discussed in https://github.com/sanni/cartreader/discussions/369

<div type='discussions-op-text'>

<sup>Originally posted by **tlwheeling** May 22, 2022</sup>
Let me start by saying _huge_ thank you for putting this project together.  I was a complete novice at soldering and this has been a great opportunity for me to improve.

I'm seeing some strange behavior on my V4 single slot reader.  I can power it on and dump 5V games/saves no problem, but the system never comes to life at 3.3V.  I can see a single green light on the Arudino, but that's about it.  Here's a picture:
[image]

Compare that to the two yellow/orange lights when the system is running at 5V:
[image]

I've actually assembled 2 complete cart readers, with 2 different sets of modules (Arduinos/LCDs/etc).  (I ordered extra parts anticipating I may screw up one).  Both behave the exact same way, and I can't find any shorts on my multimeter when testing modules in isolation, so that seems unlikely.

I've been through the other threads/troubleshooting steps/known issues but no avail.  Current thoughts:

- I read the full Mega 2560 had brownout protection so it wouldn't function at 3.3V.  Could that be an issue on the Embed?
- I opted for the no battery build.  Did I screw it up?  Here's a picture:
[image]
- Could there be newer hardware revisions on AliExpress that now behave differently?  While I ordered 2 sets of hardware via the links in the guide, AliExpress can be pretty unreliable.

Thanks in advance for any assistance!</div>

sanni, 2022-06-18:
It seems that newer Arduino Mega Pro's can't be run at 3.3V anymore. I have ordered one and will have a look as to why.

sanni, 2022-06-24:
A first clue has been found as to what could be causing the issue:

[0xyzzyx wrote](https://github.com/sanni/cartreader/discussions/369#discussioncomment-3005485):
> So my more permanent solution ended up being soldering a wire to the 3V pin for the arduino to the 3V pin of the cartridge slot on. It's a crude solution but it seems to be working for both 5V and 3V modes without any manual intervention.
> 
> [image]
> 
> This shows which two pins I wired together, but I did it on the underside of the board.
> 
> [image]
> 
> Working perfectly now. Will update if this ends up frying my board.

stewartspicer, 2022-06-27:
Hey there,

I'm having a similar issue, and I'd just like to try to verify that it's the SAME issue before I attempt the same fix.

I'm just now starting to build my reader. The only parts I have installed on the main PCB are the battery holder and the GBA reader board (with slot, headers, and switch).

5v mode, works fine. Same voltage as the 5v out on the battery. 3v mode, I get nada. Looking at the traces, I also fail to see how the power gets from the battery mount to the 3V pin there. I've read through the schematics and followed the traces, but I don't see how the 3.3V actually gets to the 3V pin

stewartspicer, 2022-06-28:
I think I'm an idjit and the power comes from the screen. Identified the voltage regulator on the LCD module, and it looks like it has a 3.3V output. I just assumed it was input, but I think I'm just gonna send it and try the above fix if I don't have voltage after the screen in installed

closed 2022-08-19 by sanni