sanni/cartreader #380: I can't record 26L6420

darleiv, 2022-06-19:
sanni did you remove support for 26l6420? because here it identifies eprom 9090 unknow and nothing else

sanni, 2022-06-19:
Did you connect it like this? 

https://raw.githubusercontent.com/sanni/cartreader/bd3eaa106b6e4cbdffbfa89a3fb5fedf3e028b17/pcb/adapters/26L6420_adapter_schematics.png

And are you using the "Flashrom Programmer"->"MX26L6420 adapter" menu option?

darleiv, 2022-06-20:
> Did you connect it like this?
> 
> https://raw.githubusercontent.com/sanni/cartreader/bd3eaa106b6e4cbdffbfa89a3fb5fedf3e028b17/pcb/adapters/26L6420_adapter_schematics.png
> 
> And are you using the "Flashrom Programmer"->"MX26L6420 adapter" menu option?

yes, who designed the adapter was you. I have it here. sop44 - friend sanni the eprom is not identical I'm using build 5.5 - when I press programmer 16 bits adp id 9090 unknow appears

darleiv, 2022-06-20:
https://ibb.co/KKXz5Wc
https://ibb.co/F7jH1NJ
https://ibb.co/37vdzrd
eu estou interessado nisto pra experiencias olhe o que criei . sera que funcionaria no seu sanni? e um conversor 8/16 bit pra eprom 26L

darleiv, 2022-06-21:
@sanni

sanni, 2022-06-23:
In flash.ino it is not swapped, in md.ino the two bytes in each word are swapped.

darleiv, 2022-06-23:
> In flash.ino it is not swapped, in md.ino the two bytes in each word are swapped.

I recorded it and after putting it on snes it didn't work why? with 8/16 bit trigger was to work hi rom no? the adapter itself I adapted to run on snes directly

darleiv, 2022-06-23:
@sanni  would this be correct?

"names":[
		"MX26L6420NC-90"
	],
        "flash_ids":[
		[  0xC2, 0x00, 0xFC, 0x22  ]
	],
	"voltage":3.3,
        "flash_size":0x800000,
	"start_addr":0x80000,
	"chip_erase_timeout":120,
        "command_set":"AMD",
	"commands":{
		"reset":[
			[ 0, 0xF0 ]
		],
		"read_identifier":[
			[ 0xAAA, 0xAA ],
			[ 0x555, 0x55 ],
			[ 0xAAA, 0x90 ]
		],
		"read_cfi":[
			[ 0xAA, 0x98 ]
		],
		"chip_erase":[
			[ 0xAAA, 0xAA ],
			[ 0x555, 0x55 ],
			[ 0xAAA, 0x80 ],
			[ 0xAAA, 0xAA ],
			[ 0x555, 0x55 ],
			[ 0xAAA, 0x10 ]
		],
		"chip_erase_wait_for":[
			[ null, null, null ],
			[ null, null, null ],
			[ null, null, null ],
			[ null, null, null ],
			[ null, null, null ],
			[ 0, 0xFFFF, 0xFFFF ]
		],
		"single_write":[
			[ 0xAAA, 0xAA ],
			[ 0x555, 0x55 ],
			[ 0xAAA, 0xA0 ],
			[ "PA", "PD" ]
		],
		"single_write_wait_for":[
			[ null, null, null ],
			[ null, null, null ],
			[ null, null, null ],
			[ null, null, null ]

sanni, 2022-06-24:
Your PCB makes no sense. It's not the Cart Readers fault.

closed 2022-06-24 by sanni

darleiv, 2022-06-24:
> Your PCB makes no sense. It's not the Cart Readers fault.
https://www.youtube.com/channel/UCGY_LrGGaLYXP1TY4PuHXiA/videos

your answer is very vague. you're not talking to a guy who doesn't understand anything about retro consoles.
look at my channel and all the projects i do
the scheme you say doesn't make sense is used in snes and it works
https://ibb.co/GPgwv4w