sanni/cartreader #388: Screen corruption due to low available memory

sanni, 2022-06-28:
When dumping Mega Drive or N64 the screen gets scrambled at the end of the dump process.

[image]

This can be fixed by disabling modules you don't need to save a little bit of SRAM.

sanni, 2022-08-19:
Was accidentally fixed with V9.5.

sanni, 2022-10-28:
Bug is back in V11.0 since you can now enable more modules without getting a compile error. 
Up to 78% dynamic memory seems to be save.

vpelletier, 2022-10-28:
This must be caused by a large local variable, like a large array.

This may not even be in code at the end of the dump, but would only become visible once the display gets refreshed.

I find [this 1024 bytes array in `readROM_MD`](https://github.com/sanni/cartreader/blob/a8a981e1373bdf9ffdacc16db14dabd66be60080/Cart_Reader/MD.ino#L1170) and [another one in `readRom_N64` when `fastcrc` is enabled](https://github.com/sanni/cartreader/blob/a8a981e1373bdf9ffdacc16db14dabd66be60080/Cart_Reader/N64.ino#L3346).

A possible quick solution could be to move them to the global ram space, which of course means less ram for the local variables of other functions, which may then smash a different part of memory...

A better but harder solution is to free more ram.

closed 2023-02-10 by sanni