sanni/cartreader #409: [SNES] Super Empire Strikes Back (USA) May have an incorrect CRC32 hash

LegendaryLinux, 2022-07-17:
My Super Empire Strikes Back (USA) cartridge consistently reproduces the following log file after 8 dumps and having cleaned the cartridge four times. The same result is produced on both 5v and 3.3v settings. Loading the resultant ROM file into an emulator successfully loads and runs the game, and the cartridge also works properly on my SuperNT.

Does anyone else have this cartridge who could dump the ROM and verify the hash?

```
OSCR HW5 V9.0

[+] Super Nintendo

[+] Super Nintendo

Searching database...

Name: SUPEREMPIRESTRI
Type: LoROM FastROM
ICs: ROM ONLY
ROM Size: 16Mbit
Banks: 64 Chips: 0
Sram Size: 0Kbit
ROM Version: 1.0
Checksum: 8FE5
 
 

[+] Read Rom

Saving to SNES/ROM/SUPEREMPIRESTRI/65/...
[*******************]
Internal Checksum...OK
CRC32... 2BF99230 -> Not found
```

PsyK0p4T, 2022-07-17:
Maybe this post would fit better in discussions than directly in issues ^^

I checked on DoM, and there is many redump confirmations for "Super Star Wars - The Empire Strikes Back" USA. Both v0 and v1. But none of them do have 2BF99230 as crc32. Can you :
1) check if you switches are all configured for SNES dumping
2) clean the cart connectors with a Qtip and some kind of alcohol. then retry ?
3) open up the cart and show us some pics of the pcb (to confirm game serial, revision, or maybe repro ?)

LegendaryLinux, 2022-07-17:
I verified the header hash (?) `8FE5` against the `snes.txt` file, and it seems to match against these two lines, with the exception of the CRC32 hash.
```
Super Star Wars - The Empire Strikes Back (USA).sfc
F45B15BE,8FE5,12,048
```

In response to your requests:  
1. I have confirmed my switches are all configured properly for SNES dumping.
2. I have been using cotton swabs (QTips) and 91% Isopropyl alcohol to clean my the cartridges.
3. I do not have a recessed star tool to open the cartridge. I will see about ordering one.

I will also move this to discussions, as requested.

closed 2022-07-17 by LegendaryLinux