sanni/cartreader #425: Problems with STL files (frame & n64sleeve)

Ancyker, 2022-07-22:
I'm having issues with getting the STL files for the frame and N64 sleeve printed. I ordered my PCBs from JLCPCB and also tried to order the 3D prints from them as well.

Here's what they said about...

**frame.stl**
Automated: https://prnt.sc/Ea2gph5ZHrud
Human reviewed: https://prnt.sc/nEG_2CumNznb

**n64sleeve.stl**
Automated: https://prnt.sc/C4l1jcY--la5
Human reviewed: https://prnt.sc/0hOJn1jjVEEA

I'm not sure if I'm doing something wrong as I'm not super familiar with 3D printing so I apologize in advance if this is some sort of user error. Also, since I'm posting anyway, what's the recommended material/method for these?

sanni, 2022-07-23:
I just fixed the frame.stl, can you please try again.

The n64sleeve.stl I can't fix, it needs 0.4mm thick walls or else the N64 cartridge won't fit over it. Just make sure you get the correct 2.5mm N64 slot, then you won't need it.

Ancyker, 2022-07-25:
They did approve it but it has a warning for the walls only being 0.8mm. I think they want 1mm, but I'll have them print it anyway and see what happens.

closed 2022-08-19 by Ancyker