sanni/cartreader #436: Can't dump Alwa's Awakening - 512kb ROM

Lo0twig, 2022-07-25:
As the title says. I can't dump the Game Alwa's Awakening. I set the correct mapper (002) but i can't choose rom size higher than 256kb.

sanni, 2022-07-25:
In line 34 of NES.ino you find the following line that defines the mapper 2:    
````
2, 3, 4, 0, 0, 0, 0, // uxrom     
````
The numbers correspond to the following settings in the same order: mapper, prglo, prghi, chrlo, chrhi, ramlo, ramhi.      

Mapper is the number of the mapper. And the -lo, -hi values define the selectable range of sizes according to these arrays:    
````
int PRG[] = {16, 32, 64, 128, 256, 512};   
int CHR[] = {0, 8, 16, 32, 64, 128, 256, 512};    
byte RAM[] = {0, 8, 16, 32};   
````
So in case of mapper 2 currently is allows a prg size between 128(3rd entry in the array, counting starts at zero) and 256(4th entry in the array, counting starts at zero again).     

So to enable 512K as selection you need to change the line to:    
````
2, 3, 5, 0, 0, 0, 0, // uxrom     
````

Lo0twig, 2022-07-25:
Thanks i'll give it a try. Danke dir Sanni <3

sanni, 2022-07-25:
[Here](https://github.com/MiSTer-devel/NES_MiSTer/issues/239#issuecomment-807611089) it is mentioned that this game unlike OG NES carts starts its PRG-RAM at 0x6000 so you might have to change the base from 0x8000 to 0x6000 too.

https://github.com/sanni/cartreader/blob/02a7b420525d8b1a983983995b4eb80b9b42d0ab/Cart_Reader/NES.ino#L2325
https://github.com/sanni/cartreader/blob/02a7b420525d8b1a983983995b4eb80b9b42d0ab/Cart_Reader/NES.ino#L2362

Lo0twig, 2022-07-25:
The Reader is giving me: CHR SIZE: 0k RAM SIZE: 0k

Lo0twig, 2022-07-25:
The creators of the game will send me the ROM.