sanni/cartreader #460: 'DDRG' was not declared in this scope updating HW3 to V9.4

hitmanmcc, 2022-08-09:
Below is a screenshot and a partial copy of the error messages (too long for GitHub). Feel free to reach out for additional information I can help with.

[image]

C:\Users\andre\OneDrive\Ambiente de Trabalho\V9.4_Portable\Arduino IDE Portable\portable\sketchbook\Cart_Reader\WS.ino: In function 'void dataOut_WS()':

WS:1321:3: error: 'DDRA' was not declared in this scope

   DDRA = 0xff;

   ^~~~

C:\Users\andre\OneDrive\Ambiente de Trabalho\V9.4_Portable\Arduino IDE Portable\portable\sketchbook\Cart_Reader\WS.ino:1321:3: note: suggested alternative: 'DDRB'

   DDRA = 0xff;

   ^~~~

   DDRB

exit status 1

'DDRG' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sanni, 2022-08-09:
Please check if you have selected Arduino Mega under Tools -> Board:

[image]

hitmanmcc, 2022-08-09:
That was it, thanks!

closed 2022-08-09 by hitmanmcc