sanni/cartreader #467: Auto naming files

Icesythe7, 2022-08-14:
currently on version 9.4, it will auto name your dumps and then if you try to flash them it cant display the file as the name has characters that it can't print or are too long(idk the issue) manually renaming the file fixes the issue.

To reproduce dump pokemon blue, then try to write it to a flashcart and it doesn't exist as an option until you manually rename the file.

sanni, 2022-08-19:
Either I'm using the Sdfat function getName() not correctly or the library has a bug.

[image]

[image]

[Arduino Sketch Download](https://github.com/sanni/cartreader/files/9381907/truncate_error.zip)

sanni, 2022-08-19:
Must be a bug in SdFat library. Truncating works in 2.1.1 as expected but not in 2.1.2 or 2.2.0
https://github.com/greiman/SdFat/releases

sanni, 2022-08-19:
Should be fixed with V9.5_Portable by increasing file name length limit and reverting to SdFat library 2.1.1.

closed 2022-08-19 by sanni