sanni/cartreader #480: Game Gear support w/Retron 3-in-1 adapter

Mythris29, 2022-08-18:
Just tested dumping a game gear game using the retron 3-in-1 adapter and it appears to work! CRC check matches dat-o-matic and I was able to play the rom in an emulator that recognizes .sms as game gear games (like pico). I just used the SMS/Mark 3 raphnet mode and had the adapter plugged into the megadrive slot. 

If code was added to match game gear crcs in that mode I think that's all that's needed.

Please let me know if I can assist in testing.

sanni, 2022-08-21:
Can you test if it works like you wanted?
Just extract SMS.ino after download and copy it to \V9.5_Portable\Arduino IDE Portable\portable\sketchbook\Cart_Reader\ and upload the sketch.
[SMS.zip](https://github.com/sanni/cartreader/files/9389732/SMS.zip)

Mythris29, 2022-08-21:
> Can you test if it works like you wanted? Just extract SMS.ino after download and copy it to \V9.5_Portable\Arduino IDE Portable\portable\sketchbook\Cart_Reader\ and upload the sketch. [SMS.zip](https://github.com/sanni/cartreader/files/9389732/SMS.zip)

Lion King which is a 512K cart worked (dumped 3 times in a row and matched the CRC in the database) but I also tried a 32K cart "Columns" and got a different CRC every time. It seems really close.

I've tested the Lion King rom and it works great the columns dump doesn't work at all but does work on a retron 5 so I think the cart is fine.

Mythris29, 2022-08-21:
[Desktop.zip](https://github.com/sanni/cartreader/files/9390056/Desktop.zip)
 Forgot to attach examples of the dumps. Lion king is great and the other 2 are the Columns dumps that kept giving me different CRCs. I tried looking at them in a Hex viewer and Lion king definitely looks more like what I would expect while the columns dump has a lot of repeating characters.

Mythris29, 2022-08-21:
[image]

sanni, 2022-08-22:
And Columns did dump fine in the "SMS/Mark 3 raphnet mode" before?

Mythris29, 2022-08-22:
I just got the game yesterday so I didn't test it when I submitted this issue. It looks like it has the same behavior in the SMS/Mark 3 raphnet mode where it dumps something with a different crc each time.

PsyK0p4T, 2022-08-24:
Different CRC everytime is strange. Try to clean the pins once more.
Then can you test this file with Columns + the Retron adapter and let us know the results ?

[SMS.zip](https://github.com/sanni/cartreader/files/9417014/SMS.zip)

Mythris29, 2022-08-24:
Yeah I thought so too the only other time I've had that was with a bad mega drive slot. So I cleaned the cart again and uploaded that new sms.zip and it works! Columns was recognized and same crc every time so it looks like whatever changed in that fixed it!

Mythris29, 2022-08-25:
@PsyK0p4T this may be unrelated, but I've also noticed that Sega Game Cards show the same behavior as the 32k GG carts using this 3-in-1 adapter. Different CRC every time I try to dump it and I've tried all the modes. I'll open a different issue for that but just wondering if what you found that fixed these GG games might apply to the Sega Game Cards as well.

The one I'm trying to dump is https://datomatic.no-intro.org/index.php?page=show_record&s=26&n=0150 Ghost House US which seems like it's a 32k card.

Thank you again for fixing the game gear dump issue!

PsyK0p4T, 2022-08-25:
Glad it fixed your GG problem. Same thing happened with retrode adapter and 32k games, I just added a condition so retron will handled same. If you have the opportunity to grab more GG games for testing, I would be interested by your results for the games I listed here : https://github.com/sanni/cartreader/discussions/452

About Ghost House, I assume you own the SMS USA retail cart. If you do, then the hashes you must get are here : https://datomatic.no-intro.org/index.php?page=show_record&s=26&n=0149
The one you posted was a beta build, so different results.
Still, retail and beta builds are all 32k. Try to redump it, and if you don't fall on CRC32 "f1f8ff2d" we'll try to modify things for SMS too.

PsyK0p4T, 2022-08-25:
Ok. Let's try with this one then. 32k GG and SMS games should be handled same now. Let me know the CRC you'll get.
[SMS.zip](https://github.com/sanni/cartreader/files/9426260/SMS.zip)

Mythris29, 2022-08-25:
I got this error trying to upload to the cart reader 

Arduino: 1.8.19 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Temp\arduino_modified_sketch_599851\SMS.ino: In function 'void readROM_SMS()':

SMS:614:132: error: expected ';' before ')' token

         sdBuffer[currByte] = readByte_SMS((raphnet_mode_sg1000 || (cartSize == 32 * 1024UL)) ?  0 : 0x8000) + currBuffer + currByte);

                                                                                                                                    ^

exit status 1

expected ';' before ')' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

PsyK0p4T, 2022-08-25:
This is what happens when I try to only use notepad from the office... lol.
This one should compile better
[SMS.zip](https://github.com/sanni/cartreader/files/9426620/SMS.zip)

Mythris29, 2022-08-25:
The version uploaded correctly but I'm still getting different CRCs every time. To make sure I'm not messing up testing with this should I be using SMS retrode mode to dump this? And I can confirm the card works on my Retron5 so I believe it's in decent shape at least since the Retron dumps the roms before playing them.

Also I'm selecting 32k for the rom size and on the screen after that I get:

SMS/GG Header Not Found

Name: CCCCCCCC
Selected Size 32KB

Press Button...

I'll attach the rom dump I'm getting. This one had a CRC of 2A9C050D but again it's different every time I try it. 
[187.zip](https://github.com/sanni/cartreader/files/9428778/187.zip)

PsyK0p4T, 2022-08-26:
Ok so, if you have to select the rom size manually, then it means that the expected "TMR SEGA" string was not found/read by the dumper. And it could mean two things : 
- the string really doesn't exists in ROM (very rare, but it can happen. I already had the case on Sega Mark III games)
- or the string can't be read correctly. This happens when the cartridge pins are not clean enough OR that the dumper just don't see anything (no cartridge inserted, or bad pinout/access to the cart)

By looking at your file, there is no real data inside, it's nearly blank. So it's not a corrupted/incomplete dump. The dumper just didn't read data. If your cartridge was inserted, I would say that the mode you used (SMS retrode ?) is maybe not appropriate to use for dumping SMS with Retron adapter.

What result do you have when using SMS Raphnet mode ?
Did you successfully dump another SMS game with Retron adapter ? If so, which settings did you use ?

Mythris29, 2022-08-26:
> What result do you have when using SMS Raphnet mode ?
Same results. I'll put another copy of a bad ghost house dump here though just in case.

[194.zip](https://github.com/sanni/cartreader/files/9435196/194.zip)

> Did you successfully dump another SMS game with Retron adapter ? If so, which settings did you use ?
Yes I have been able to dump Great Baseball just fine (and it matches a database CRC) using the same adapter in SMS/Mark 3 raphnet mode.

***After I was done writing this I did another dump with great baseball (SMS) using the retron adapter no issues then I immediately put in the sega game card and it found the header and dumped ghost house correctly! It matched the CRC and everything...I have no been able to repeat those results again. I've tried scrubbing the pins on both the adapter and card and adapter card slot multiple times but I can't seem to repeat the successful dump results. I have another game card Choplifter that I just got that I tried as well with no luck.

Very confusing.
[TMR SEGA.txt](https://github.com/sanni/cartreader/files/9435203/TMR.SEGA.txt)

PsyK0p4T, 2022-08-27:
If the original topic of this thread (dumping GG games with Retron adapter) is done, then maybe we should close it and start another discussion focused on SMS dumping with Retron.

If you sucessfully dumped ghost house (at least once, with correct crc etc) then you confirmed that the ino code is working, and also the mode to use with your Retron adapter. Two good news.

Now, if you can't reproduce the good dump again, you can try different things :
- gently play on the cartridge or the adapter, maybe one pin is not perfectly in contact somewhere
- reformat your SD card, using exfat
- try a different SD card
- blow/clean the different connectors (the cart reader's one, the adapter one...)
and try again Ghost House :-/

Mythris29, 2022-08-27:
I agree this ticket has gone beyond the scope of fixing game gear games. I'll do some more testing then open another issue if I think we can make some progress. I'm also ordering a sega card catcher that I'm hoping will help test this.

Thank you for your help again!

closed 2022-08-27 by Mythris29

PsyK0p4T, 2022-08-27:
@sanni I let you check once more the latest SMS.ino posted ([here](https://github.com/sanni/cartreader/issues/480#issuecomment-1227500838)) and commit it if everything ok ?

sanni, 2022-08-28:
Sure go ahead.

PsyK0p4T, 2022-08-31:
@Mythris29 I also ordered a Sega card catcher to test it with Raphnet and Retrode adapters. I'll report Sega Cards dumping results with both.

If someone is able to find a Retron 3-in-1 sega adapter (for at decent price...) I would be also interested to get one :)

FatBeard, 2024-09-02:
I think i'm having a similar issue to this one when dumping columns. My sonic 1 and 2 game gear carts dump fine, but for colunns it's unable to find the header. I'm unable to confirm if the cartridge itself works in a game gear but i have no reason to believe it doesn't. Here is the log below from the OSCR. Any thoughts? thanks.
`OSCR HW5 V14.4

[+] SMS/GG/MIII/SG-1000

[+] GameGear Retrode

[+] Read ROM

[+] 32 KB

Selected Size: 32KB
Press Button...

SMS/GG header not found
 
Name: UNKNOWN
32KB
 
Press Button...

Saving to GG/ROM/UNKNOWN/186/...
ROM Size: 32KB
[*******************]
CRC32... 55B1810D -> Not found`

andy-miles, 2024-09-03:
> I think i'm having a similar issue to this one when dumping columns. My sonic 1 and 2 game gear carts dump fine, but for colunns it's unable to find the header. I'm unable to confirm if the [...]

Hmm.  After inspecting a valid dump of a US/EU Columns GG rom dump, the "TMR SEGA" header is present, but not mirrored.  I'll need to get a hold of a GG adapter to test/repro.  Otherwise, my hunch is that it's the auto cart size detection algorithm defined in `getCartInfo_SMS()`.  I've ran into similar issues myself with SMS carts, which is the reason I added the manual ROM size selection menu option override as a work-around.  Long term, I think the cart-size detection logic needs a re-design/re-write (e.g., a checksum + database lookup).

The CRC32 of the rom dump I'm looking at is `83FA26D9`.   Perhaps try using an emulator to load your dump to see if it works?  If not, try cleaning the cart contacts, manually selecting 32 KB ROM size, and trying again?

In the meantime, I'll put in an order for the PCB and connector to build myself an adapter so I can see if I can reproduce the issue.

FatBeard, 2024-09-04:
I tried it again there and i got a successful dump. However it didn't detect the header. I selected 32kb as you mentioned in the menu and the dump was identified with the correct crc in the sanni. So i'm happy with this. 

However, the funny thing was I had done all this before, i had even gone as far as reflowing the solder in the carts pcb itself and using ipa. One thing I didn't mention yesterday is i'm using the game gear adapter outlined here https://github.com/sanni/cartreader/discussions/168 with the retrosix cart slot. 

Either way, i'm happy with the succesful dump. Thanks for your help