sanni/cartreader #481: Unable to dump SA1 games with HW5.

splash5, 2022-08-19:
In six slot adapter, snes slot pin 25 connects to D40, but in HW4 or HW3 it connects to CIC_RESET.
After rewiring pin 25 back to CIC_RESET, I can dump SA1 games with HW5 flawlessly.
Is there any reason that pin 25 connects to D40?
[image]

sanni, 2022-08-19:
You're absolutely right. My mistake. I uploaded a fixed Six Slot Adapter Rev2.

Summary: CIC Reset (marked CR on PCB) was not connected between CIC adapter and SNES slot. Instead D40 was connected both to D40 and CIC Reset. Two cuts to traces need to be made and D40 and CR need to be rewired on Rev1.

[image]

kamalawala, 2022-08-20:
Wait, so can SA1 games be dumped with HW5 now? How to build wiki page still says the snesCIC adapter hasn't been made yet.

Mythris29, 2022-08-20:
I'm not sure if this is helpful or not but I built the v3 CIC adapter and put it on a rev1 6 slot adapter board and have been able to dump the 1 SA game I have (Super Mario RPG). I've had to reboot 5-6 times in some cases but it does seem to have worked or at least the CRC check passed. I haven't had time to actually try it in an emulator yet.

splash5, 2022-08-21:
@kamalawala I build CIC adapter using universal PCB. You can check [https://github.com/sanni/cartreader/tree/master/hardware/cic_adapter](url) which sanni uploaded last month.

@Mythris29 After I built CIC adapter I first test "Super Mario RPG" but it just not recognized even I reboot about 10 times. After correcting wiring around CIC_RESET all my SA1 games can be correctly detected and dumped without problem (both checksum and CRC32 passed).

TCorgard82, 2022-08-23:
So this was really helpful, I just checked my Rev 1 board and they have this issue, so that might be my issue with SA-1 games. thank you, will update if it fixes my issue.

TCorgard82, 2022-08-23:
[image]

So the mod got me to read SA-1 games, but I get checksum and CRC32 error, but I took a close up of the PCB and marked the traces to be cut.

closed 2022-08-24 by sanni

Mythris29, 2022-09-17:
Not to necro this topic but has anybody ordered revision 2 and tested it yet? I just got mine a few days ago and I can't dump my SA1 game but I was able to on revision 1 it just took 4-5 times of rebooting for it to work.

I'm testing with Super Mario RPG. I'm able to read non SA1 games just fine but after 20 reboots I still haven't been able to read Super Mario RPG once with my revision 2 six slot on HW5.

TCorgard82, 2022-09-18:
So I think I have rev 1 6 slot, and I had to modify the pcb to get the
right connections to the CIC PCB.

KytorIndustries, 2023-02-11:
Thank you this was very helpful, I was able to get CIC working on HW5 rev 1.