sanni/cartreader #483: 12Mbit Shining Force Version

Mythris29, 2022-08-20:
I attempted to leave an edit on the md.txt file for approval but I don't think I did it right and/or don't have permissions.

There is a 12mbit version of the Sega Genesis Shining Force that isn't in the md.txt file and I happen to have that version of the cart. It's confirmed on no-intro (or at least others have this CRC version as well)

https://datomatic.no-intro.org/index.php?page=show_record&s=32&n=1174

Shining Force (USA).md
E0594ABE

If this isn't the right place to add things like this I'll gladly put it somewhere else. Just trying to help with the project with testing as much as I can :)

sanni, 2022-08-21:
Added

sanni, 2022-08-26:
Removed again until it's listed as a good dump on no-intro

closed 2022-08-26 by sanni