sanni/cartreader #486: Famicom Wars won't write save dump

sanni, 2022-08-20:
### Discussed in https://github.com/sanni/cartreader/discussions/485

<div type='discussions-op-text'>

<sup>Originally posted by **Mythris29** August 20, 2022</sup>
So I think the dump is working for sram on my copy of Famicom Wars because it has info in it if I look at it in a hex editor but I can't seem to write the file back to the cartridge. 

I don't get any errors during the process so I'm unsure what the issue is. I have 2 famicom carts and both dump the carts just fine so I don't think the slot and or ver5 cart reader is the issue but I only have a sample size of 2 and only a sample of 1 for this sram issue so I'm unsure if this is a hw5 issue or an issue just with my setup

I'm attaching a copy of the sram dump is case it helps.

[RAM.zip](https://github.com/sanni/cartreader/files/9386408/RAM.zip)
.</div>

PsyK0p4T, 2022-08-25:
Don't know if I can help on anything here, tests or so... 
I've a Famicom Wars cartridge here, and hw v3 + FC adapter.

sanni, 2022-08-25:
Would be great if you could play the game until there is a save on the cart. Then dump the save. Then overwrite a copy of the save with 0xFF on all bytes and flash that 0xFF copy back to erase the save in the cartridge. Confirm there are no write errors and that the save is gone when playing the game and then write the backed up save back to the cartridge and confirm that the save is there again when playing the game.

PsyK0p4T, 2022-08-25:
Understood your plan. Unfortunately I have the cart, but not a Famicom to play it.
I can eventually : 
- dump the current save, whatever is on it
- Then prepare a 0xFF save. Flash it to the cart.
- Try to dump the save from cart, confirm it is 0xFF
- Flash back the original save.
- Dump it again to confirm the content ?

Mythris29, 2022-08-25:
I'm happy to do this I would just need to know how to prepare a 0xFF save. I have a retron that I can create and test the save with.

Mythris29, 2022-08-25:
[RAM.zip](https://github.com/sanni/cartreader/files/9428767/RAM.zip)

If it's easier to just modify this file and send it back to me that's fine as well (just dumped this save from my cart). But I'd love to learn how to be more helpful testing this so either way is fine with me :)

sanni, 2022-08-26:
> I can eventually

That would work too

PsyK0p4T, 2022-08-26:
@Mythris29 I grabbed your bin file to get the expected RAM size, then made another one fully filled with 0xFF value.
As listed above, the goal would be to flash it to the cart, then read the cart again to confirm 0xFF save it was written fine.
Then upload the original bin save you did. Dump it again, and confirm than old save was written fine.

@sanni Ok i will try it and let you know the results

Mythris29, 2022-08-26:
Writing the 0xFF file didn't erase my save game. (I assume that's what it was supposed to do)

PsyK0p4T, 2022-08-29:
Finally took the time to test this. Here are my results : 
- used Famicom Wars (HVC-FW) on hw3 + Famicom-to-SNES adapter + software v9.6
- switches set to NES mode (right left left left)
- game is automatically recognized and info loaded from database
- dumping rom worked
- dumping sram worked : "RAM.bin"
- created a 8KB file "blank.bin" only filled with 0xFF and put it on my sd
- flashed that blank file to the cartridge SRAM
- dumped again the SRAM
- the file obtained was matching "blank.bin"
- then flashed the original save back to SRAM
- dumped again the SRAM
- the file obtained was matching "RAM.bin"

So everything worked as expected from my side. SRAM backup and restore operations had no problem. 

Also adding my files here, even if they will not be really useful... can you maybe retry once more by using the blank.bin file @Mythris29  ?
[saves.zip](https://github.com/sanni/cartreader/files/9448378/saves.zip)

Mythris29, 2022-09-05:
Still troubleshooting this but I may need to get my hands on some original famicom hardware to troubleshoot this. 

1. Backed up current save on cartridge
2. Wrote blank save to cartridge
3. Backed up save again then compared hex files
4. First backup has data as expected second backup is blank or all 0xFF as expected
5. Put Famicom wars into Retron 5 to play and the save is gone
6. Put Backup save with data back onto the cartridge and did another dump to confirm it's on the cartridge. No save game data is on the cartridge though when I try to play it on the retron. If I look at the CRCs they don't match even though the save data looks similar via hex editor view.

I've disabled save states and save backups on the Retron and wipes it's directories but something still isn't right and I can't rule out the retron might be messing with something but the CRCs not matching is very strange.
[Saves.zip](https://github.com/sanni/cartreader/files/9492283/Saves.zip)

sanni, 2022-09-06:
Judging from PsyK0p4Ts and your test it seems to me like either the battery in the cartridge is empty or the Retron is messing with the save somehow.

closed 2022-09-06 by sanni

Mythris29, 2022-09-07:
Just to leave some useful notes. I got a Famicom to NES adapter today to test this out with and it seems to work perfectly. Even with save states and save backups turned off the retron5 seems to still do strange things with saves and can't be trusted to test saving issues.