sanni/cartreader #502: Release 9.5 Screen Corruption on 3V HW5 R3

WindyBison, 2022-08-30:
Software
- Release 9.5 (then downgraded to 9.4 to solve the issue)
- Portable method used for flashing the Adrino via the "How to Flash"
- Windows OS used for the Adrino flashing operations

Hardware (PCB from JLPCB and Parts from "What to order" links)
- 2 slot configuration (GBX, N64)
- HW5 R3
- N64 Adaptor
- KeyStudio MEGA PRO 2560 
- SI5351 clock generator
- MKS MINI12864 V3 LCD

Please excuse me for I am new to GitHub. If this is not the right place to report issues, I apologize. 

5volt worked perfectly, no screen issues and was able to read the ROMs of Tetris (Gameboy) successfully 

3volt, however, showed corruption on the letter "b". (attached 2 photos) and would have a checksum error when trying to look up GBA Games from the database file (Tested 3 cartridges, all failed)

[image]

[image]

I decided to re-flash the Adrino with Version 9.4 with NES module disabled and Megadrive disabled in the code

Refreshing the Adrino with Version 9.4 solved the corruption and I was to dump GameBoy, GBA, and N64 cartridges no problem.

Could this be an issue only in release 9.5?

sanni, 2022-08-31:
Can you try what happens when you reduce the file name character limit in 9.5 again to the same as it was in 9.4? 
The setting is found in [Cart_Reader.ino in line 363](https://github.com/sanni/cartreader/blob/master/Cart_Reader/Cart_Reader.ino#L363)

Change:

````
// File browser
#define FILENAME_LENGTH 100
#define FILEPATH_LENGTH 132
#define FILEOPTS_LENGTH 20
````

to:

````
// File browser
#define FILENAME_LENGTH 32
#define FILEPATH_LENGTH 64
#define FILEOPTS_LENGTH 20
````

WindyBison, 2022-08-31:
```
// File browser
#define FILENAME_LENGTH 32
#define FILEPATH_LENGTH 64
#define FILEOPTS_LENGTH 20
```
The above change you suggested resolved the screen letter "b" corruption on version 9.5 while 3V was selected. I was able to dump Bomberman Tournament (GBA) and also read the save 

Tested the ROM on mGBA and it worked no problem.

sanni, 2022-09-06:
If you find the time it would be interesting to see when exactly the issue appears. E.g. you could try only changing either #define FILENAME_LENGTH 32 or #define FILEPATH_LENGTH 64 to see which line is causing it. And then once the culprit is found slowly increase the size until it breaks.

Also interesting would be if using shorter cables from the display to the main PCB changes anything.

WindyBison, 2022-09-13:
I had to re-download the latest v9.5 portable release as I deleted the previous one due to cleanup.

Not sure if there were any changes to v9.5 between now and when I opened the issue, but I cannot replicate the issue anymore. (I also did not change anything hardware wise)

To try and replicate the base issue to test out changing FILENAME_LENGTH or FILEPATH_LENGTH, I flashed the Arduino with the latest v9.5 release (on Sept 12) with no edits on FILENAME_LENGTH or FILEPATH_LENGTH and only edited the code to define HW5.

I did not get any screen corruption issues on the letter "b" as before with 3volt.

I also tried disabling all modules but n64, GBX, flash, and flash16 and flashing the Arduino again.

I still was not able to replicate the issue, everything worked just fine for dumping a n64 cart, gba cart, gba save, and Gameboy cart.

closed 2022-09-18 by sanni