sanni/cartreader #522: Sega 32x missing

flyingclimber, 2022-09-18:
Is there any reason not to have 32x checksums in either md.txt or a dedicated 32x.txt? I tested dumping Star Wars Arcade and it passed internal checksum but not CRC as expected. I can submit a PR if we are ok to add it

sanni, 2022-09-18:
I don't have any 32x carts, can you test if [this code](https://github.com/sanni/cartreader/blob/master/Cart_Reader/MD.ino#L1316) selects the 32x database correctly?

[image]

flyingclimber, 2022-09-19:
No, it still tries to use md.txt even though it shouldn't. I'll add some debug statements and see what's going on later this week

flyingclimber, 2022-10-16:
This still happens with V10.2. The calculated checksum is correct and the same as 32x.txt but for some reason it's still using md.txt

sanni, 2022-10-16:
I changed the code slightly try today's update.

flyingclimber, 2022-10-16:
That works perfect. No issues

Why does the address for readWord_MD() have to be divided by 2? Does it have to be aligned to only read from an even address?

Thank you!

sanni, 2022-10-16:
The Mega Drive always reads two bytes per given address. 

So while viewing the ROM in an hex editor in Windows you would have:
Address - Data
0x0 - 0xAA
0x1 - 0xBB
0x2 - 0xCC
0x3 - 0xDD

The MD reads like this:
Address - Data
0x0 - 0xBBAA
0x1 - 0xDDCC

So whenever you reference an address based on a ROM file on your PC you have to read an even address divided by 2.

It also has a reversed byte order compared to a Windows PC.

````
PC:
Little Endian
 
Nintendo:
Wii = Big Endian
Wii-U = Big Endian
Switch = Little Endian
Gamecube = Big Endian
Gameboy = Little Endian
Gameboy Color = ???
Gameboy Advance = Little Endian
DS = Little Endian
3DS = ???
Nintendo 64 = Big Endian
NES = Little Endian
SNES = Little Endian
Virtual Boy = Little Endian
 
Playstation:
Playstation 1 = Little Endian
Playstation 2 = Little Endian
Playstation 3 = Big Endian
PS Portable = Little Endian
 
XBOX:
XBOX = ???
XBOX360 = Big Endian
 
Sega:
Master System = Little Endian
Mega Drive/Genesis = Big Endian
Game Gear = Little Endian
Sega CD = Big Endian
32X = Big Endian
Saturn = Big Endian
Dreamcast = Little Endian
````

So it's easy to get confused.

closed 2022-10-16 by sanni