sanni/cartreader #527: Issue dumping Sega Master System Carts

ccfman2004, 2022-09-22:
Every time I try to dump a Master System cart I get a different CRC value which is not found. I have the HW5 R3 board with the Rev1 Six Slot Board.  The one cart I tried to dump was Mono Poly which is in the database on the SD card but the CRC value is different every time I dump. When I open what is dumped it does have some plaintext that is definitely from the game. 

I select SMS and then the first option and click to read rom but each dump yields a different CRC value. 

I have no issues dumping SNES games including SA1 games and as far as I can tell the solder joints look good so I'm not quite sure what is wrong.

Also the WIKI NEEDS to be updated with instructions form using the HW5 boards since there are a number of differences between HW3 and HW5.

sanni, 2022-09-22:
Monopoly is one of those special games with a ton of additional chips, it most likely will need some extra code to be dumped correctly.

[image]

List of more games that use Sega's 315-5208, 315-5235 or 315-5365 memory paging chips:
Action Fighter	
After Burner	
Alex Kidd in Miracle World	
Altered Beast	
Astro Warrior	
Bubble Bobble	
Casino Games	
Chase HQ	
Galaxy Force	
Gangster Town	
Ghostbusters	
Golden Axe	
Hang-On/Astro Warrior	
Kenseiden	 
Mercs	
Monopoly	
Operation Wolf	
Penguin Land	
Phantasy Star	
R-Type	
Rastan	
Sagaia	
Space Gun	
Space Harrier	
Spellcaster	
The Ninja	
Wonder Boy III: The Dragon's Trap	
World Grand Prix	
Source: https://www.smspower.org/Development/SMSPagingChips

ccfman2004, 2022-09-22:
Of course the 3 games I have are on that list. Although the 3rd game is Hang Out/Safari Hunt not Hang Out/Astro Warrior.

Can I send you one of the games or send it to someone who can help develop the code? I bought the games to make sure the cart reader I was building was 100% working. Had no idea there were games that couldn’t be dumped.

PsyK0p4T, 2022-10-20:
I looked a bit around this. Not sure if it's only due to these games having paging chips embedded, but more that it's hard to detect/predict the exact rom size for SMS/GG.

The current SMS.ino tries to read the info in rom header first, by checking the checksum area value. This often gives a correct size, but not always.
Then makes an additionnal check to see where the "TMR SEGA" string is repeated. This will eventually correct/adjust the rom size and will work for the very large majority of games (Ex : "Double Dragon" or "Super Monaco GP" need this).

But this adjustment sometimes oversizes other games, like the ones you're discussing here. As a temporary solution, I commented the "size fix" part. Try to use the SMS.ino attached and let us know the results. I dumped "Operation Wolf" or "Altered Beast" correctly with it.
But be warned that by using this ino, some other games will now have their size badly detected.

Hope that someone here can figure out an universal trick...

[SMS.zip](https://github.com/sanni/cartreader/files/9827885/SMS.zip)

PsyK0p4T, 2023-02-13:
Just deleted my previous old post here, where I reported issues to dump "Galaxy Force".
I retried today after cleaning the cart once again. And it dumped fine with the current code (v12.2), and all the different adapters tested so far (raphnet, retrode, retron).

The PCB only includes two chips : the MPR-12566A (game) and a 315-5235 (paging).
So "just" a paging chip is not a problem. But Monopoly or others games which contain more chips, could be...

Ancyker, 2023-03-23:
[image]

PsyK0p4T, 2023-04-11:
@sanni we can probably close this issue ?
Paging chips seem to be out of cause, since I dumped Galaxy Force and Ancyker did Monopoly too.

If @ccfman2004 comes back on this topic, try to update to v12.5 and follow the quick guide posted above to get the expected crc.

closed 2023-04-11 by sanni