sanni/cartreader #535: 6 Slot Adapter V2

Mythris29, 2022-09-27:
I ordered some of the 6-slot adapter boards v2 and after making 2 of them both don't let me dump CiC required (SA1) games. I tried to look through the differences on the schematics but it's over my head...I guess it's possible that I messed up making 2 of them but my v1 board works fine. I'm also able to dump non SA1 super nintendo games just fine.

So far I haven't seen anybody who can confirm they've gotten a v2 board and it works so I just wanted to open this up and make sure there isn't an issue with the design of the board itself.

Mythris29, 2022-09-27:
Oh I'm on HWv5

ccfman2004, 2022-09-28:
I ordered some rev2 boards and are waiting for them to arrive. Once they do and I can test one I will update.

Mythris29, 2022-09-28:
> I ordered some rev2 boards and are waiting for them to arrive. Once they do and I can test one I will update.

Awesome, I'm not ruling out that I'm the problem here but I'd like to at least see some others attempt to build it and see if it works or not :)