sanni/cartreader #536: Better Way to Select NES/Famicom Game

ccfman2004, 2022-09-28:
There needs to be a better way of either the cart reader to properly read all games or a faster way to select a game that is near the end of a very long list of game like in the "S" section. It takes way too long to scroll down to get to Super Mario Bros. 3 Japan for example.

sanni, 2022-09-28:
Can you update to the latest version from today and post the oscr log file entry on the SD after dumping Super Mario Bros. 3 Japan like so:

````
OSCR HW5 V10.0

[+] NES/Famicom

Searching database
for 34E9CD01... 

TwinBee (Japan).nes
MAPPER:   87
PRG SIZE: 32K
CHR SIZE: 16K
RAM SIZE: 0K
Rotate to Change
Press to Select

NES CART READER

CURRENT SETTINGS

MAPPER:   87
PRG SIZE: 32K
CHR SIZE: 16K
RAM SIZE: 0K

[+] Read iNES Rom

Saving to NES/ROM/TwinBee/254/...
[*******************]
CRC32... 985B1D05 -> TwinBee (Japan).nes
````

ccfman2004, 2022-09-29:
I updated the Arduino and SD card files and it is not able to dump correctly at all.  Every time I select NES/Famicom it displays a different CRC value after I manually scroll to the game it does not dump correctly at all and each time I dump it is a different CRC value.

I tried to dump a GBA game that I've dumped successfully before and even it isn't being dumped correctly although it does display the correct game name before dumping it. 

It's doing some funky things now. I tried the USA version of Excitebike and it said it was the European version at first then after dumping it said Japan, USA. 

While scrolling to get to Super Mario Bros. 3 Japan I had a blank screen. 

[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/9670959/OSCR_LOG.txt)

ccfman2004, 2022-09-30:
It seems my copy of Super Mario Bros. 3 Japan may be borked but I have other issues with the 10.0 release. 

My copy of "Rad Racer USA" is mis-identified as "Highway Star JAP" as the value that shows up initially is for Highway Star but after it dumps the game the CRC32 value matches "Rad Racer US". Not sure what's going on there.

Also it can't properly dump "Sesame Street ABC's USA" which is 2 games in one and it gives me an error "No data found at 0x8000
Using manual selection" before letting me pick the letter.

Also sometimes it doesn't dump a game properly the first time and it does the second time. 

Here is another log:
[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/9681620/OSCR_LOG.txt)

sanni, 2022-09-30:
Can you try changing 0x8000 to 0xC000 in this line https://github.com/sanni/cartreader/blob/master/Cart_Reader/NES.ino#L436
And then report what happens to Sesame Street ABC's and Super Mario Bros. 3.

ccfman2004, 2022-10-03:
As for my Famicom copy of Super Mario Bros. 3, there is something wrong with it as I can't even dump it with v9.5 now. I have another copy on the way as well as a Famicom to NES adapter to make absolute sure my current copy has stopped working.  Every time I select to read this game I get a different CRC value. I also have a few more Famicom games on the way for further testing. I was able to properly dump that game originally but not anymore. 

I still have to manually select Sesame Street ABC's as it still can't find it automatically. 

[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/9695220/OSCR_LOG.txt)

ccfman2004, 2022-10-06:
The Famicom to NES adapter arrived and my copy of Super Mario Bros 3 works fine and property dumps using the Famicom to NES adapter on the NES slot. It still makes me manually select the game as the ID value it initially reads keeps changing but using the proper mapping values it dumps fine. Not sure why the Famicom slot isn't working anymore but I got my order of REV 2 Six Slot adapter PCBs in so I'm going to start fresh and update. 

[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/9726065/OSCR_LOG.txt)

sanni, 2022-10-06:
I have just added auto detection for (some) MMC3 mapper games. You need to update both the nes.txt database and the firmware on the OSCR.
Super Mario Bros. 3 should now be correctly detected.

But same as with Rad Racer/Highway Star it won't detect the region though. That usually doesn't matter but you can use the selection wheel to move one right to the Japanese version.

ccfman2004, 2022-10-07:
Now it properly detects Super Mario Bros. 3 Japan. Another Famicom game, Dragon Ball - Shen Long no Nazo, it cannot find as the two values don't match what's in the nes.txt file but manually editing the mapping data does correctly dump the game. 

I hope one day we will never have to manually edit the mapping data. 

Thanks

[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/9730496/OSCR_LOG.txt)

ccfman2004, 2022-10-07:
I fixed the Famicom slot. I tested continuity, AGAIN, and apparently the slot's Chip Enable pin didn't have enough solder to create a proper bond so it couldn't properly read the cart which absolutely explains why the final CRC32 value was never the same twice as it wasn't properly turning chips on. OOPS.  LOL.  I only figured this out by checking the actual slot pin rather than the PBC's via point for CE. 

I cannot believe I made such a newbie error. LOL.