sanni/cartreader #548: Issue dumping N64 eeprom (tried Mario 64 and Banjo Kazooie)

ThreePinkApples, 2022-10-08:
I have an Open Source Cartridge Reader V3-ALTER (bought through the build service) and have updated the firmware to v9.5
I can dump the game ROM fine for both of these, but when I try to dump the save it seems stuck on "Saving eep....". I have tried with the 1K pull-up resistor switch both off and on.
If I boot these cartridges on my N64 the saves are there and can be loaded, so they're not corrupted.

txt files from the ROM dumps:
```
OSCR HW3 V9.5

Searching database...

N64 Cartridge Info

Name: BanjoKazooie
ID: NBKP Size: 16MB
Save: 4K Eeprom
Version: 1.0
 

Saving to N64/ROM/BanjoKazooie/15/...
CRC32... 525899C9 -> Banjo-Kazooie (Europe) (En,Fr,De).z64
Done (214s)
```

and 

```
OSCR HW3 V9.5

Searching database...

N64 Cartridge Info

Name: SUPERMARIO64
ID: NSMP Size: 8MB
Save: 4K Eeprom
Version: 1.0
 

Saving to N64/ROM/SUPERMARIO64/9/...
CRC32... 03048DE6 -> Super Mario 64 (Europe) (En,Fr,De).z64
Done (109s)
```

sanni, 2022-10-08:
Have you installed the Clock Generator?

[image]

And set the switches correctly?

[image]

And also checked that "#define clockgen_installed" is not removed from Cart_Reader.ino?

https://github.com/sanni/cartreader/blob/master/Cart_Reader/Cart_Reader.ino#L119

ThreePinkApples, 2022-10-08:
I did not have the CLK1 switch turned on, haven't found anything that said what use it had. But turning it on fixed it!
I was using this guide: https://savethehero.builders/blogs/features-notes/how-to-use and it doesn't mention the need for the CLK1, but I do see it mentioned here: https://github.com/sanni/cartreader/wiki/Overview

sanni, 2022-10-08:
The old firmware version had "#define clockgen_installed" disabled for HW3 and didn't need the Clock Generator installed and the CLK1 switch had to be set to OFF instead. But people had issues with some carts.

So I wrote a new way that needs "#define clockgen_installed" enabled, needs the Clock Generator installed and needs the CLK1 switch set to ON. This is also what HW4/5 uses.

closed 2022-10-08 by sanni

sanni, 2022-10-08:
The next update will have the switches displayed in the menu

[image]