sanni/cartreader #552: Sometimes the dumped ROM gets saved to the root directory of the SD card instead of the correct folder

sanni, 2022-10-09:
The Cart Reader seems to fail at creating the game specific numbered folder and instead just saves to the root dir.

PsyK0p4T, 2022-10-10:
Hey Sanni. 
Dunno if this is exactly what is happening to you, but I can confirm this happens for some games having special characters in their rom internal name. They can't be used / wrote as a correct folder name, so the rom is created at SD root. Then CRC check operations also fail cause the rom can't be found in the expected dir.
[image]
[image]
[image]

Reproduced with all these JPN SNES games so far : 
Parodius
Kentouou world champion
kirby super deluxe
Sanrio smash ball
Dragon ball Z Super Butoden 2 and 3
Momotarou Dentetsu
Super donkey dong 1 and 2

sanni, 2022-10-10:
I think your issue was a buffer overflow of romName[] that I already fixed in newest release.

````
OSCR HW5 V10.2

[+] Super Nintendo/SFC

[+] SNES/SFC cartridge

Searching database......
Found

Title: SIMULATION PROYAKYUU
Revision: 0
Type: LoROM SlowROM
ICs: ROM SAVE
ROM Size: 2 MB (64 banks)
Save Size: 8 KB
Checksum: 231C

[+] Read ROM

Saving to SNES/ROM/SIMULATION PROYAKYUU/306/...
[*******************]
Checksum... 231C -> OK
CRC32... 24C9B924 -> Simulation Pro Yakyuu (Japan).sfc
````

The current issue doesn't display any error, the file is just in the wrong dir, even gets correctly renamed to no-intro and everything. 
Ofc whenever I want to reproduce the issue to investigate it doesn't happen.

PsyK0p4T, 2022-10-12:
Confirmed, you fixed this with romName[] update. My bad !

sanni, 2022-10-14:
Upgrading to Arduino IDE 2.0.0 with the newest libraries out of the library manager seems to have fixed the issue for me. The low memory warning when compiling with the standard modules enabled is gone now too. Could be related.

closed 2022-10-14 by sanni