sanni/cartreader #577: [janitorial] flashRepro_N64: sectorSize and bufferSize may be used uninitialised

vpelletier, 2022-10-22:
I am trying to clear all build warnings, and for this one I have no idea how to resolve it.

The function's general structure is:
```c
void flashRepro_N64() {
  unsigned long sectorSize;
  byte bufferSize;

  if (cartSize != 0) {
    // does not initialise sectorSize and bufferSize
  } else {
    // initialises sectorSize and bufferSize by user input
  }
  // this part uses sectorSize and bufferSize
}
```

So it looks like they are only initialised in the exceptional case (unknown flashcart ?). I suspect this is a "true" bug, but I have no such cart so I cannot check.

vpelletier, 2022-10-22:
There is another warning about an uninitialised variable in one of the functions called from there: `writeFlashBuffer_N64`.

In that case it is a bit clearer that this should not be an issue: the variable would be uninitialised if `bufferSize` is 0, which seem unlikely to happen (...if the caller takes care to never provide 0, but the caller has bigger issues).

sanni, 2022-10-22:
You can initialize them to a value of zero if it removes the warning. In the case that they are needed they get new values anyway in the two switch cases.

vpelletier, 2022-10-22:
Do they ? The `switch` blocks are only for the case where `cartSize == 0` (the `else` branch). The only lines outside of that branch which touch these variables are:
```c
void flashRepro_N64() {
  unsigned long sectorSize;
  byte bufferSize;
// ...
      } else if (bufferSize == 0) {
        writeFlashrom_N64(sectorSize);
      } else {
        writeFlashBuffer_N64(sectorSize, bufferSize);
      }
```

Also, setting to zero will cause another issue in `writeFlashBuffer_N64`. So I think there is more to fix here, but I do not know enough to tell exactly how it should be fixed.

sanni, 2022-10-22:
I see, maybe like this:

````
// ...
} else if ((bufferSize == 0) && (sectorSize != 0))  {
        writeFlashrom_N64(sectorSize);
      } else if (sectorSize != 0) {
        writeFlashBuffer_N64(sectorSize, bufferSize);
      }
else{
print_Error(F("sectorSize not set"), true);
}
````

closed 2022-10-23 by vpelletier