sanni/cartreader #578: [janitorial] getCartInfo_SNES reads 1kB and does nothing with it

vpelletier, 2022-10-22:
From `SNES.ino`:
```c
  uint16_t c = 0;
  uint16_t currByte = 0;
  byte buffer[1024] = { 0 };
  PORTL = 192;
  while (c < 1024) {
    PORTF = (currByte & 0xFF);
    PORTK = ((currByte >> 8) & 0xFF);
    // a few NOPs
    buffer[c] = PINC;
    c++;
    currByte++;
  }
```

`buffer` is never used outside of this loop. The first easy cleanup step is to drop it, but this loop has side effects: it writes to ports F and K. Are these writes expected to have any effect on the gamecart state ? Otherwise I can just drop this entire loop.

sanni, 2022-10-22:
I think I was trying to improve reading the SNES cartridge with SA1 special chip. You can delete the whole loop. 
This all stems from a time where I still had plenty room in the Atmega available. Better to cut it down as much as possible now and then worry about stuff breaking later on. 👍

vpelletier, 2022-10-22:
Thanks, applied in my fix-warnings commit. Closing here, I intend to submit a merge request later.

closed 2022-10-22 by vpelletier