sanni/cartreader #619: Arduino bootloader does not erase the last supposedly-usable flash page

vpelletier, 2022-11-09:
This is not really a bug specific to this project, but which should be worked around to prevent some head-scratching.

The Arduino bootloader used in the chip targeted by this project has a bug which causes it to never erase the last flash page. As a result, all writes to that last page can only ever clear bits, never set them. In practice, this means that out of the intended maximum usable flash space of 248kiB, 8kiB are usable only once, and then will forever cause verification failure (and incorrect operation) when the programmed image is larger than 240kiB.

I see a few possible workarounds:
- fix the bootloader. This is hardly within the scope of this project. There is [an outstanding merge request on the corresponding Arduino repository since 2020 addressing this bug](https://github.com/arduino/Arduino-stk500v2-bootloader/pull/1). Visibly, the submitter did not sign the contributor agreement, leaving us to deal with this bug.
- get the Arduino build environment to deduce 8kiB from the usable flash space. From what I can see, the Arduino build environment is extremely rigid (no real way to set compiler options, so I do not expect a way to tweak a chip model's linker mapping), so this seems unlikely to lead anywhere.
- add 8kiB of bloat at a fixed address in the resulting image. I think this must be possible in a few inline assembly lines, but so far I am failing to get gcc to comply. The idea would be to fill that bloat with 0xff bytes, so `avrdude` would ignore them (it looks like it has a feature which never even sends trailing 0xffff words to the chip, so this would take no extra programming time). This way, if the firmware crosses that boundary, the build will fail rather than produce an image that cannot be reliably used.

Maybe someone on this project has the know-how to implement a working solution ?

Ancyker, 2022-12-01:
Fixed in PR via https://github.com/arduino/Arduino-stk500v2-bootloader/pull/2

Ancyker, 2022-12-01:
Since I'm not sure when/if ever the Arduino team will approve the PR, here's a quick how-to on flashing it yourself that is specific to the OSCR:

First up, you need an additional Arduino (or an ICSP capable of programming an ATmega2560). Flash it with the ArudinoISP (it's included in every IDE installation, click File > Examples, it should be in there) before beginning.

You can shortcut this next part by using the latest OSCR release (V10.2 at the time of posting). Make sure the Arduino IDE is closed and navigate to `V10.2_Portable\Arduino IDE Portable\hardware\arduino\avr\` and open `boards.txt` in your plain-text editor of choice (not Word). Find this:

```ini
mega.menu.cpu.atmega2560.build.mcu=atmega2560
mega.menu.cpu.atmega2560.build.board=AVR_MEGA2560

## Arduino Mega w/ ATmega1280
## -------------------------
```
...and right above the "## Arduino Mega w/..." comment add:
```ini
## Arduino Mega w/ ATmega2560 [Fixed]
## -------------------------
mega.menu.cpu.atmega2560fix=ATmega2560 (Mega 2560) [Fixed]

mega.menu.cpu.atmega2560fix.upload.protocol=wiring
mega.menu.cpu.atmega2560fix.upload.maximum_size=253952
mega.menu.cpu.atmega2560fix.upload.speed=115200

mega.menu.cpu.atmega2560fix.bootloader.high_fuses=0xD8
mega.menu.cpu.atmega2560fix.bootloader.extended_fuses=0xFD
mega.menu.cpu.atmega2560fix.bootloader.file=stk500v2/stk500boot_v2_mega2560-fix.hex

mega.menu.cpu.atmega2560fix.build.mcu=atmega2560
mega.menu.cpu.atmega2560fix.build.board=AVR_MEGA2560
```

Then save the file. You should have something like this:
```ini
mega.menu.cpu.atmega2560.build.mcu=atmega2560
mega.menu.cpu.atmega2560.build.board=AVR_MEGA2560

## Arduino Mega w/ ATmega2560 [Fixed]
## -------------------------
mega.menu.cpu.atmega2560fix=ATmega2560 (Mega 2560) [Fixed]

mega.menu.cpu.atmega2560fix.upload.protocol=wiring
mega.menu.cpu.atmega2560fix.upload.maximum_size=253952
mega.menu.cpu.atmega2560fix.upload.speed=115200

mega.menu.cpu.atmega2560fix.bootloader.high_fuses=0xD8
mega.menu.cpu.atmega2560fix.bootloader.extended_fuses=0xFD
mega.menu.cpu.atmega2560fix.bootloader.file=stk500v2/stk500boot_v2_mega2560-fix.hex

mega.menu.cpu.atmega2560fix.build.mcu=atmega2560
mega.menu.cpu.atmega2560fix.build.board=AVR_MEGA2560

## Arduino Mega w/ ATmega1280
## -------------------------
```

Download [this file](https://raw.githubusercontent.com/Ancyker/Arduino-stk500v2-bootloader/master/goodHexFiles/stk500boot_v2_mega2560.hex) into `V10.2_Portable\Arduino IDE Portable\hardware\arduino\avr\bootloaders\stk500v2\` and make sure it is named `stk500boot_v2_mega2560-fix.hex`.

Open the Arduino IDE. When you go to select your processor a new option should appear, choose it:
[image]

Wire up your Arduinos. You need to connect the SCK, MOSI, and MISO pins to each other. Then connect pin 10 of the programmer to the reset pin of the OSCR. For more details on wiring [see the official Arduino guide](https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP). Make sure to only connect USB to the Arduino doing the programming, you should have something like this:
[image]

Make sure you've selected "Arduino as ISP" and the COM port of the Arduino that will be doing the programming and click "Burn Bootloader":
[image]

It should only take a few seconds. If you are updating an existing OSCR the display will go blank once flashing is complete. After that just disconnect the jumper wires and reconnect your cart reader to a PC and update it as normal. You only need to do this once per Arduino as the bootloader is never overwritten (except via this method, of course).

@sanni could simplify this a bit by including the hex file and boards.txt change in the next release. Not sure if that's something they want to do or not.

Note: This isn't actually needed at the moment for this project. If the project ever does use any of the last 8k of space it will fail to update on subsequent attempts at flashing new updates. This can be mitigated by making the program not use any of the last 8k of space. This works because if the program does not use it then avrdude does not try to verify it. The data location will still be invalid but because nothing is trying to read it that means it doesn't affect anything. 

This means the project can use the last 8k of space even though some people who cannot update can't use it, they'll just have to disable something so the last 8k isn't used. If @sanni edits the releases' boards.txt and modifies `mega.menu.cpu.atmega2560.upload.maximum_size` to be 8k less that likely will prevent people from uploading invalid code (unless they choose the fixed version when they do not have the fixed version, of course).

closed 2023-02-10 by sanni