sanni/cartreader #632: [NES] 72-pin slot too shallow for some carts

nsx0r, 2022-11-21:
There are reports of contact issues with some (early?) NES PCB versions that have shorter contact pins, and I can confirm this myself.

bad contact "no data found" when fully inserted:
[image]
same game (newer PCB?), good dump:
[image]

It seems like the 72-pin slot isn't deep enough to allow proper contact to these shorter pins.
I'm trying to find a different slot than the one recommended on the wiki, but they all look the same and rarely specify their exact dimensions.

Meanwhile you can *force* the pins to make contact but I don't like pushing/yanking the PCB in the slot and the results are random.

Stanton54, 2022-12-13:
I had an issue like this with my dragon warrior and when I took off the back screws it worked for my v5.

MattroidPrime, 2022-12-17:
I too had this issue with my Dragon Warrior cart along with a couple of others. While the others I was able to dump by holding down on the cart with a bit of force, The DW cart just wouldn't have it, even when removed from the shell. I ended up getting an adapter board and pushing the back row of pins up using a pair of needle nose pliers just to the edge of the socket before soldering it to the board (I pushed all the back pins up even though they're not all necessary for this particular game which is all I needed it for but hey, I was on a roll). 

It still took a bit of fiddling but my v5 finally auto-recognized the cart and I eventually managed to get successful ROM and SRAM dumps. 

If I do another full build at some point, I might just push those pins up from the start unless another socket can be found by then.

[image]

Cephiros, 2023-02-04:
I did this on my second build, i cut the stands off so the connector was flush with the board and pushed all the back pins up to sticking up about 1mm above the top. Reads everything without an issue. If you add this as a step in the build process wiki, you can probably close this, otherwise I wouldnt count on a fixed slot from the famiclone producers.

closed 2023-02-10 by sanni

PeteCologne, 2024-08-01:
The EUR Version of Megaman 1 PAL B also doesn't work. It's an UNROM-08 pcb https://nescartdb.com/profile/view/1416/mega-man - the only game of my collection. Rest works great.