sanni/cartreader #633: NP GB Flashcart

dannygonzalez0861, 2022-11-22:
Hi,

I am having an issue flashing to a Nintendo Power Gameboy memory Cartridge.

I purchased the cartridge with only one large ROM file so I had to start from scratch. I am trying to load Pokemon Blue alone w/o the menu but when I try and flash the ROM or MAP files, the Cart reader reboots.

When I "Erase Flash" I get the warning message and press to confirm, the next page shows Erasing... but no "OK" and then asks to press the button.

Next, I Blankcheck which immediately says "Blankcheck...OK" and asks to press the button to continue..

Pressing Read Flash and Read Mapping, generate corresponding files in the NP folder of the SD card.

Please help me, literally purchased this cart and built your reader for each other.

Thanks!

Daniel G.

sanni, 2022-11-22:
Try updating to [V11.4](https://github.com/sanni/cartreader/archive/refs/heads/master.zip)

dannygonzalez0861, 2022-11-22:
> Try updating to [V11.4](https://github.com/sanni/cartreader/archive/refs/heads/master.zip) And if that doesn't fix it you can disable some modules that you don't need and see if that resolves the issue.

Hello,

So I am writing this for future reference but to save those who don't care, yes this V11.4 did indeed work with the NP GB Memory Cart and the menu allowed me to select the appropriate files on the SD card.

Well to dive into what I did, my Arduino is modified (Fuse removed) so I simply left it installed on the main board and powered it on via USB as normal. I then followed the steps to flash a new Arduino as on the instructions but the files above do not contain the libraries nor portable IDE so I simply installed a fresh copy of IDE and used the libraries folder found in the prior (V10) version. Note I had to transfer the files from the download location to the Arduino IDE files in the My Documents Folder, again the instructions page was extremely helpful here in finding the location of the folders in question (The bottom of the page describes IDE standalone install vs. Portable). After IDE updated the libraries and added the Cart_.... file to IDE I defined v5, installed drivers, and selected the board and port verbatim to the instructions. I did remove a few unused modules as well but my prior version also excluded these modules so I do not think that was the issue. Once flashed the Arduino and the screen rebooted and like magic, I was able to erase the flash, Blankcheck...OK and then when writing to the GBMC instead of rebooting as before, I am welcomed to a file browser and allowed to select the appropriate ROM and MAP files.

All is well in my world again and I want to formally thank you Sanni for the assistance, the project, and the joy it has been to put together and get working.

Thanks!

Daniel G.

closed 2022-11-22 by dannygonzalez0861