sanni/cartreader #653: Sega Genesis/Master Drive /Cart - D40 Pin Issue

dannygonzalez0861, 2022-12-09:
Hi Sanni,

I think I might have just discovered an issue with the six-slot PCB Rev. 3.

I have been trying, w/o success, for the past week to get Mega Drive/Genesis Carts to read but the moment I plug one in and try to turn the reader on, it just stays off. I disconnect the game and try again and the reader powers on like usual. If the reader is powered on and I try to connect the game, it just shorts out and turns off.

I had a bit of free time and decided to see if I could find a short on my PCB but did not find a defect from my soldering. Well after a few hours of testing/probing for shorts and looking over your schematics/ Sega PCBs I physically have, I noticed that D40 which has continuity to VCC goes connected to the /cart pin of the genesis port which on all of my games is actually a ground pin.

I confirmed the result by applying a piece of tape to the /cart pin to avoid getting VCC to the pin as it shares ground on the game itself and the reader stays on as well as successfully pulls the ROM and passes the checksum.

Can you confirm if this is a one-off or a design issue?

dannygonzalez0861, 2022-12-09:
[image]
[image]
[image]

Ancyker, 2022-12-09:
D40 should not have continuity to VCC.

Ancyker, 2022-12-09:
I was looking over the PCB layouts and the only place I see D40 ever getting close enough to VCC that it could accidentally be bridged is on the CIC adapter:

[image]

dannygonzalez0861, 2022-12-13:
I can't seem to find where the short on this board is. It is not on the CIC unfortunately, I disconnected the board as a quick test and the short still exists. I half-built a second unit over the weekend with the spare parts I had and it does not present these issues. It must be a one-off problem with this PCB.

Thanks for the help!

closed 2022-12-13 by dannygonzalez0861

Ancyker, 2022-12-13:
I think I had a similar issue with my first one. It was caused by a bridge
under one of the headers. You could try reflowing all the headers to see if
it helps.

dannygonzalez0861, 2022-12-30:
Well, i though i would share what I found was the issue after all. I have no idea how this happened but the header for the SNES_CIC PCB was shorted almost completely. Not sure if I had a few too many that day but here is what I found when I started to disect the PCB, hope you all can laugh at my pain LMAO
[image]
[image]

@Ancyker your help was much appreciated, all of your ideas were indeed acurate and I just needed to fins exactly where the isse was.

Thanks!