sanni/cartreader #664: MD Zero Wing (E) ROM Read Issue

jaffa225, 2022-12-20:
After trying to get in touch to see if you had a solution on the original Arduino thread, I, luckily, was able to solve this myself.  So that it may be better noticed, and added to the project code, I'm posting it here too.

I'm sorry if diff/patch files aren't the usual format for patches here.  It's the first change I've suggested to a github project, but should be simple to incorporate as it's only a few lines. Thanks for everything!

The cartridge was reading as 4 Mbit, until I added another case for "Zero Wing (E)", and now it happily reads the full 8 Mbit, with this patch (renamed to .txt to be uploaded here):
[V10.2_Portable.edited.txt](https://github.com/sanni/cartreader/files/10265514/V10.2_Portable.edited.txt)

Here's the log file created by the patched V10.2 (portable) software while successfully reading the ROM: 
[ZEROWING.txt](https://github.com/sanni/cartreader/files/10265524/ZEROWING.txt)

This solution was originally posted at [https://forum.arduino.cc/t/rom-reader-for-super-nintendo-super-famicom-game-cartridges/155260/1570?u=jaffa225man](url)

Thanks again!

sanni, 2022-12-20:
Thanks alot, added it to the code. 👍

closed 2022-12-20 by sanni

jaffa225, 2022-12-21:
> Thanks alot, added it to the code. +1

You're quite welcome, and thank you for adding it! :)