sanni/cartreader #679: Unable to read LRG NES cart

retrovik, 2023-01-11:
I am unable to read the Limited Run Games SCAT NES cart. It seems to read fine at first (finds the entry in the database), but then gives an error on the CRC check at the end. The file created is not usable. Using HW V3 and Firmware beta V12.0 (I also tried 10.2). I am guessing the fact that this is not an actual mapper 1 board but an FPGA simulation of one is the issue. I am not sure if this is a bug or a feature request. Just FYI.

[SCATSpecialC.txt](https://github.com/sanni/cartreader/files/10394768/SCATSpecialC.txt)

[image]

[image]

nsx0r, 2023-01-11:
Hi, could you join Discord to run some tests? https://discord.gg/YVxv7BnzXt

Cephiros, 2023-01-11:
Looking at that cart, seems like alot going on for an mmc1 game. I wouldnt be surprised at all if limited run built some anti-dumping code into that FPGA that can detect if its plugged into an actual NES, or if something is just reading and probing the address and data lines.

nsx0r, 2023-01-13:
I think I have a workaround, come back to Discord ;D

retrovik, 2023-01-15:
@nsx0r Your workaround worked perfectly. I just had to manually edit the generated header in a hex editor since it was in ascii. After that, it works great now. Thanks!

[image]

[image]

nsx0r, 2023-01-17:
this issue can now be closed, as it's related to NES 2.0 not supported at the moment https://github.com/sanni/cartreader/issues/442

closed 2023-01-18 by retrovik