sanni/cartreader #685: Display Backlight Not Working

breyell, 2023-01-16:
Admittedly, this may be beyond the scope of this project but figured it might be something that someone may have stumbled upon and knows how to fix. I've put together a Sanni v5 hw3 and after it was together, the backlight on the display didn't work Everything else worked great, including dumping data. Swapping the arduino with a new one resolved the issue. Do you know what I may have done wrong or what to check on the arduino?

sanni, 2023-01-16:
The LCD's background LED is controlled via the Adafruit Neopixel library through Arduino pin 13. It could be that pin 13 is not soldered correctly or maybe shorted to a neighboring pin on the first Arduino.

breyell, 2023-01-17:
I'll check this out within the next few days. Thank you for the guidance.

breyell, 2023-01-19:
Pin 13 isn't bridged with neighboring pins and I'm getting continuity from the arduino to the IDC connector. Any other ideas?

breyell, 2023-01-19:
The D13 pin (26) on the ATmega2560 wasn't soldered properly. Reflowed it and now it works great. Thanks for the help!

Diagram for others that have this issue:
https://components101.com/sites/default/files/component_pin/ATMega2560-Pinout.png

closed 2023-01-19 by breyell