sanni/cartreader #692: NES: Issues dumping games with 8K PRG like Galaxian (Japan)

sanni, 2023-01-29:
### Discussed in https://github.com/sanni/cartreader/discussions/691

Both the automatic mapper setup using the iNES header from nes.txt and the manual mapper setup don't allow a PRG size smaller than 16K and therefore the OSCR doesn't dump games with 8K PRG like Galaxian correctly.

The resulting ROM looks like this: iNES header - 8K PRG - 8K PRG(mirror) - 8K CHR

Ancyker, 2023-03-23:
Just wanted to note here I'm completely redoing how NES.ino works, so please don't change it significantly XD

[image]

(Sorry to post this 3 times, just wanted to note it on all of the NES issues)

nsx0r, 2025-01-22:
should be fixed now
https://github.com/sanni/cartreader/commit/caccb84d6695972e651d711485279f757db43949

closed 2025-01-22 by nsx0r