sanni/cartreader #699: SNES: Dumping issue in v8.0 onward (v3 hardware)

Jenetrix, 2023-02-02:
SNES carts (Original, Repro, and SF Memory) will refuse to read correctly on software v8.0 and higher when configured for V3 hardware (with snescic + clockgen enabled.) Issue is present regardless if clockgen is enabled or disabled when compiling.

Pin PC7 (SNES pin D7) is pulled high when attempting to read cartridges and results in an invalid CRC (CB9E instead of 4B9E for Mario Paint.) Repro carts will fail to detect at all and prompt for manual settings input, and SF Memory carts will fail with a "Hirom All Timeout" error. 

Last working software version that did not have this issue was v7.3 
Cartridges Tested: Mario Paint (US), Mario All Stars + Mario World (US, Repro), SF Memory

Jenetrix, 2023-02-02:
The issue does not appear to be with the Arduino/ATMega itself, as swapping it out with the same software uploaded results in identical symptoms. Downgrading back to 7.3 from 12.0Beta/10.2 resolved the issue for me.

sanni, 2023-02-03:
Can you try to overwrite single *.ino files in 7.3 with the ones from 8.0 and test after each replaced file if the issue appears?
That way we could limit it to one file and then in this file replace all changed lines one by one until we find the culprit.

PsyK0p4T, 2023-02-03:
All the SNES retail carts you tried failed ? or just few titles ?
I don't have repros, but I redumped some originals across the last versions, with hw v3.
PCB with SA1/SFX chips too, and [SF memory carts](https://datomatic.no-intro.org/index.php?page=show_record&s=49&n=1832).

Jenetrix, 2023-02-03:
> Can you try to overwrite single *.ino files in 7.3 with the ones from 8.0 and test after each replaced file if the issue appears? That way we could limit it to one file and then in this file replace all changed lines one by one until we find the culprit.

I tried swapping 8.0 *.ino files into the 7.3 folder's sketchbook and I was unable to replicate the issue in testing. Re-uploading 8.0 back onto the Arduino however seemed to work this time. Updating again to 12.1 also seemed to work correctly (as did 12.0 beta). No changes were made to the 8.0 or 12.0 options since my last testing. (HW3, clockgen_calibration, clockgen_installed) I'm thinking it may possibly have been caused by updating straight from 6.4 to 12.0 which I had done prior to attempting to dump the aforementioned SNES carts. Bringing it back to 7.3 first before updating to 12.1 seems to have resolved it for the moment.

That said now that I can get to the point it dumps, the SF Memory will still give an invalid checksum on the flash when verifying. (NP/TOKIMEKIMEMORIA/17, 7D91 != 47C9, CRC32 4638E7D3 Not Found.) This happens regardless of cable or power supply used. 

> All the SNES retail carts you tried failed ? or just few titles ? I don't have repros, but I redumped some originals across the last versions, with hw v3. PCB with SA1/SFX chips too, and [SF memory carts](https://datomatic.no-intro.org/index.php?page=show_record&s=49&n=1832).

These three carts are the only SNES carts I have, apart from a pair of SGBs.

PsyK0p4T, 2023-02-05:
About the SF Memory : 
- if you got a CRC error message for the "Menu" rom dumping, it will be normal. Info about the games flashed on the cartridge are written into the Menu. It has to be manually cleaned to get the virgin "Menu" rom.
- About Tokimeki Memorial, there is one report in DoM coming from a NP cart. What rom size and rom version do you have? Expected values are rev1, and 4194304 bytes.

Jenetrix, 2023-02-05:
The .bin files that got dumped to \NP\\
[image]
[image]

Inside of \NP\TOKIMEKIMEMORIA there are several numbered sub-directories and the roms dumped to those folders have calculated checksums that mismatch. (7D91 & E8F0)

PsyK0p4T, 2023-02-05:
Your .bin files are looking good. On the screenshots you posted : "CRC32 AE9F3602"
That is the expected CRC for [Tokimeki Memorial - Densetsu no Ki no Shita de (Japan) (Rev 1)](https://datomatic.no-intro.org/index.php?page=show_record&s=49&n=3089) so... matching known good dumps. Maybe the header checksum failed but your rom looks good.

Jenetrix, 2023-02-09:
Ive been reading and writing the SF Memory for the last few days with various versions of the software and playing with the clockgen calibration a bit and have been unable to replicate the issue again. Closing issue.