sanni/cartreader #713: Flashback for Genesis not detected

PsychoFox11, 2023-02-18:
Is this where I should report missing titles? Apologies if not.
CRC32 23a9616d is Flashback: Quest for Identity, US version, but comes up as not found.
Also listed here: https://segaretro.org/List_of_Mega_Drive_ROM_dumps

PsyK0p4T, 2023-02-18:
Hi. 
can you confirm that the rom you obtained got a size of 1572864 ?

sanni, 2023-02-18:
According to the [dat-o-matic](https://datomatic.no-intro.org/index.php?page=show_record&s=32&n=0504) no-intro overdumps it on purpose even though the internal ROM header declares it as a 1.5MB ROM.

[image]

If you trim the 2MB no-intro ROM to the correct 1.5MB you also get the CRC32 of 23A9616D.

PsychoFox11, 2023-02-18:
Does this mean it can't be detected properly by the reader, or can the info in the files be adjusted to make it detect/dump correctly? The ROM plays perfectly fine and that is the CRC32 I get, but I haven't compared ito an official no-intro, I'm just checking out my own. Or will the reader only use no intro CRC32s even if they aren't quite right? Learning how all this ties together.

sanni, 2023-02-18:
Can you replace MD.ino with this version, upload and then test again?
It forces the OSCR to overdump the cartridge, curious what happens.

[MD.zip](https://github.com/sanni/cartreader/files/10775334/MD.zip)

PsychoFox11, 2023-02-19:
That caused it to identify correctly, with a CRC32 of 9FEFB1AB

sanni, 2023-02-19:
Ok, could you do one more test and temporarily edit the size of Flashback to 4MB in MD.ino by replacing 0x200000 with 0x400000 in line 777:

[image]

Dumping will now ofc produce an error. But I would be interested in how the ROM looks like in an hex editor like [Nexus HxD](https://mh-nexus.de/en/hxd/) at Offset(h) 00200000

[image]

PsyK0p4T, 2023-02-19:
Thanks for these additions in code sanni :) That's what I wanted to investigate since a while (since we confirmed "Shining Force" correct size). We used the same approach, making an big overdump to check where the final padding stops, and where repeated data starts.

I don't see this fix as an overdump, but more a "full chip dump". Until now we followed the info given in header, which sometimes just gives the rom/game size, but not the chip size.

also reduced the code a bit, and merged previous fixes with yours : 
[MD.zip](https://github.com/sanni/cartreader/files/10777116/MD.zip)

closed 2023-02-19 by sanni

PsychoFox11, 2023-02-19:
Just checking, did you still need me to check those Sanni, or does this being closed mean you have the info you need?

sanni, 2023-02-19:
Got all the info I needed, thanks.